OrcaSlicer/OrcaSlicer #1679: PA Pattern: Extra cube being added

R4toLoko, 2023-08-04:
First of all, thank you so much for everybody involved on this this PR.

I have been following that PR and did notice the little cube being added.
It looks like the PR was merged with it.
It would be cool for the non-beta release to remove that.

[image]

Thank you

TexZeTech, 2023-08-04:
I found I had to add a second cube to the right side just so I can have [KAMP](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging)  probe the area correctly. 

So ideally it would still pull the dimensions of the final code generated and remove that cube from it.

R4toLoko, 2023-08-04:
I use [Frix-x](https://github.com/Frix-x/klippain/tree/main) instead and indeed, now that you mentioned it and to actually print and no mesh is performed and Mainsail shows a small cube which I believe is that little cube I mentioned.

[image]

Also, it does not respected the profile's speed.
It hurts seeing the printer at 30mm/s when it was supposed to be 105mm/s

[image]

Ohh that is why and no idea why that 30mm/s is there:

[image]

Even if I change that, the letters are locked to 30mm/s
[image]

I hope this will be fixed, it should not alter the speed/acceleration whatsoever.

R4toLoko, 2023-09-12:
The little cube is still showing and the mesh is working as expected.
The slower first layer is a minor thing so I am closing this.

closed 2023-09-12 by R4toLoko