OrcaSlicer/OrcaSlicer #1680: Quality issue with "Single wall on top surface" with 1.6.4 beta

Hammer57, 2023-08-04:
1.6.4 removes too many inner walls with this setting on. Leading to a rougher surface.

Left 1.6.4 Beta, Right 1.6.3

"Single wall on top surface" On
[image]
"Single wall on top surface" Off
[image]
"Single wall on top surface" On
[image]

STL
[Ghost.zip](https://github.com/SoftFever/OrcaSlicer/files/12256926/Ghost.zip)

Noisyfox, 2023-08-04:
Well unless you fix your printer's top layer quality, the extra walls won't really help. Your model just happened to have your upper layers cover the lower walls. If you print models like the benchy then your top surfaces will have low quality regardless.

Hammer57, 2023-08-04:
Don´t think that the problem is my printers top layer. 
These are from 1.6.3. Definitely better as the top of the Ghost with 1.6.4.
[image]
[image]
[image]

Not yet printed a Benchy or Calicube with 1.6.4

Noisyfox, 2023-08-04:
Try classic wall generator and see if the same issue presents with Single wall on top surface turned on.

Hammer57, 2023-08-04:
Better but still not like 1.6.3.
[image]

Another thing: Could it be that the setting did not work as intended in 1.6.3 with Arachne? 
With Classic "one wall" On the preview looks the same in 1.6.4 and in 1.6.3. 
With Arachne there was only a minor difference between On/Off in 1.6.3 (screenshots in first post) in comparison to 1.6.4

locki-cz, 2023-08-04:
@Hammer57 Problem is you dont understand how it works, It set one perimeter on all top surfaces, in some cases like this on it will look every time worse thena more perimeters, you should use there modifiers if it is possbile like in prusaslicer and set it just on last layer or dont use it on that object at all OR use concetric top layer fill. Concentric looks better on round objects :)

Noisyfox, 2023-08-04:
Looks like you have too much overlap between the infill and the walls, could you try decreasing it? Or maybe change the wall order to infill first?

Hammer57, 2023-08-04:
@locki-cz I do understand the concept, I was just irritated, that this was a non-issue with the previous version of OrcaSlicer. But if it was not really working (as confirmed by Noisefox) explains it. Thanks for the tip with concentric top layer fill. Totally forgot about that.

@Noisyfox Thx, I´ll try that.

Hammer57, 2023-08-04:
> Looks like you have too much overlap between the infill and the walls, could you try decreasing it? Or maybe change the wall order to infill first so the wall can cover the overlap?

Made no real difference. Left is with Infill/Outer/Inner, right is with overlap 5% instead of the default 15%

[image]

SoftFever, 2023-08-04:
Hi
I understand that this thread is already closed, but I still wanted to provide some comments on this matter :)

This issue has actually been persistent in the previous version of "single wall on top" as well. It didn't work well for models with slanted top surfaces, like the ghost model you mentioned. Personally, I tend to turn off this feature when printing such models.

You might not have noticed these artifacts because the 1.6.3 version didn't have the real "single wall on top" feature for the Arachne engine. It's a new implementation by @Noisyfox in the 1.6.4 beta.

Anyway, I made a small tweak to this feature. I'll release a new beta-2 build later, which will also include other bug fixes reported by users.

For comparisons:
Current effect:
[image]

After tweaked:
[image]

Hammer57, 2023-08-04:
Thanks for the info. Looking forward to the second beta. Looks way better in the second pic. 

BTW: Big thanks to you and everyone involved in making this awesome piece of software. 👍

igiannakas, 2023-08-05:
I've played around with the code on this - I'd recommend to increase the multiplier slightly - I'm using 3 instead of 2 as I find that while this gets rid of most of the ugliness, there are some edge cases that would benefit from a little more "room" for the nozzle to lay down the one line infill. For example, the top walls of a benchy.

igiannakas, 2023-08-08:
> Just noticed that BambuStudio has their own impl of the "Single wall on top surfaces" for Arachne. Not sure if it's better quality than us.

No it suffers from the same issue and the code is not as clean as orca. I’ve submitted a PR to resolve the same issue on the BL slicer.

closed 2023-11-22 by github-actions[bot]