OrcaSlicer/OrcaSlicer #1682: Temperature Run Away

Newcal, 2023-08-04:
**Describe the bug**
After changing the Start Gcode in My Klipper Profile or using RatRig Vcore 3.1 profile, the temperature displayed is 0 (from first layer to the last layer) - This morning after sending the g-code to the rating, I had a violent temp runaway - Not sure if it related to the bug...

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
[image]
Renamed JPEG for force upload but it is a 3MF

**To Reproduce**
the use of the vcore500mm printer profile or the change of the start gcode of My Klipper printer profile

**Expected behavior**
The display of the temp shows 0

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**
RatRig Score 500

**Desktop (please complete the following information):**
 - OS: Mac
 - Version 1.6.3
Thank you

SoftFever, 2023-08-04:
I don't have access to your model.

It displays 0 is most likely you set nozzle and bed temperature in print_start macro.
In that case OrcaSlicer won't know the actual temp you set in the macro.

closed 2023-11-22 by github-actions[bot]

streetgt, 2023-12-28:
@SoftFever can we have a fix on this? Using custom start G-Code show temp as 0. Are you able to port this feature from SuperSlicer for example?

Best regards