OrcaSlicer/OrcaSlicer #1685: Filaments cannot be deleted

PrintModz, 2023-08-04:
**Describe the bug**
I had already noted it, but even in the new version no filaments can be deleted. In addition, specially created materials cannot be deleted either. This does not work via the slicer and also not via the configuration folder.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Create filament and try to delete

**Expected behavior**
It should be possible to delete your own materials when you no longer need them.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
any

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

Grasshopper333, 2023-08-04:
I'm currently on version 1.6.4-beta, for me deleting profiles works without any problems. It also works on 1.6.3
[image]

SoftFever, 2023-08-04:
> I'm currently on version 1.6.4-beta, for me deleting profiles works without any problems. It also works on 1.6.3 [image]

Same here

kylek29, 2023-08-04:
I had this issue as well -- on my end, I believe I had to uncheck the option to File > Preferences > "Auto Sync User Presets (Printer/Filament/Process)" option. I believe when I originally troubleshot it, I determined it was a sync issue where the Bambu server would just rewrite the deleted profile.

closed 2023-11-22 by github-actions[bot]