OrcaSlicer/OrcaSlicer #1686: Wrong behavior when using "Force cooling for overhang and bridge"

Grasshopper333, 2023-08-04:
**Describe the bug**
When using the "Force cooling for overhang and bridge" option the fan randomly doesn't turn off for a few layers.

**3mf File for This Bug**
[WrongFanSpeed.zip](https://github.com/SoftFever/OrcaSlicer/files/12260731/WrongFanSpeed.zip)

**To Reproduce**
Activate the "Force cooling for overhang and bridge" option.

**Expected behavior**
The fan should turn off right after the overhang 

**Screenshots**
Current behavior on version 1.6.4-beta
[image]

Expected behavior on version 1.6.3
[image]

**Desktop**
 - OS: Windows
 - Version:  Windows 10 Home

SoftFever, 2023-08-04:
Thanks for reporting it.
it should have been fixed in Beta 2

Grasshopper333, 2023-08-04:
It looks to be working now.
Thank you for the fast fix.

closed 2023-08-04 by Grasshopper333