OrcaSlicer/OrcaSlicer #1690: Merge PrusaLink & Connect into Orcaslicer & Prusa MK4 profiles

rovili, 2023-08-04:
**Is your feature request related to a problem? Please describe.**
Not related to a problem.

**Describe the solution you'd like**
It would be great if PrusaLink and PrusaConnect features could be merged into Orcaslicer. I prefer Orcaslicer over PrusaSlicer because of major features such as single perimeter on top layers.

It would also be great to include the Prusa MK4 Profiles - especially the Input Shaper profiles.

SoftFever, 2023-08-05:
I will take a look.

Graloth, 2023-09-02:
I would love this feature as well, mainly because I want to switch to Orca Slicer, and currently there is no MK4 profiles available

SoftFever, 2023-09-02:
Hey guys, the lastest nightly build has full support for PrusaLink and MK3.
[image]

I have tested it with MK4IS, let me how well it works for you.
You can download it here:
https://github.com/SoftFever/OrcaSlicer/actions/runs/6058229688

[image]

hliebscher, 2023-09-02:
Prusa works but all other networks Switch after selecting to MKS

jhenley01, 2023-09-04:
I can upload files using the Print button but if I switch to the device tab everything shows NA. Confirmed Prusalink works as expected in a normal browser window. I'm running the Mac_arm64 version.

[image]

rovili, 2023-09-05:
Same here, uploading the gcode is functional, but the device tap shows N/A.

closed 2023-12-12 by github-actions[bot]

MacBoyPro98, 2024-08-13:
Was this issue ever fixed/a solution found for Mac_arm64 version? I'm getting a `401 - Unauthorized` error when using the device tab:
[image]

**Note:** I am trying to use this on my Mk3s+ with PrusaLink running on a Pi Zero W over the GPIO port
**OrcaSlicer Version:** 2.1.1

> I can upload files using the Print button but if I switch to the device tab everything shows NA. Confirmed Prusalink works as expected in a normal browser window. I'm running the Mac_arm64 version.
> 
> ![Screenshot 2023-09-04 at 10 08 34 AM](https://private-user-images.githubusercontent.com/56179818/265495673-72c0d871-e158-4666-ab84-428de4486dad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NTg3MjIsIm5iZiI6MTcyMzU1ODQyMiwicGF0aCI6Ii81NjE3OTgxOC8yNjU0OTU2NzMtNzJjMGQ4NzEtZTE1OC00NjY2LWFiODQtNDI4ZGU0NDg2ZGFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDE0MTM0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmODBjMDdiNThhYjE3ZGNmOTgwOTA1MzY0YjY1NzdhNDExZjVkMThmOTdmZDJkZjkwODIzMWJkY2YyYzE5OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Gy38Wyc8F1RX0HVGMEPrSa_FTWgwvBhhb-6S_4DRjTg)