OrcaSlicer/OrcaSlicer #1693: Printer Icon Application Crash Issue

rohanrohan, 2023-08-05:
**Describe the bug**
A clear and concise description of what the bug is.

Clicking the icon for the printer in the top left throws a error: "Unhandled unknown exception; terminating the application". The icon is right beneath the home button.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

Click the 3D printer icon, in the Printer Section.

**Expected behavior**
A clear and concise description of what you expected to happen.

Not sure. On Bambu Studio it brings up Custom Printer Setup.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Printer model**

Ender 3 Pro

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

1.6.4 Beta
Windows 11

SoftFever, 2023-08-05:
Thanks reporting the issue.
I can confirm the issue.

SoftFever, 2023-08-10:
I won't call it a proper fix.
But I will disabled the button as it's not useful in its current state anyway.

closed 2023-08-13 by SoftFever