OrcaSlicer/OrcaSlicer #1697: When maximized Orca Slicer window extends over the Taskbar and into the Left Monitor

DIYPERSPECTIVE, 2023-08-05:
**Describe the bug**
Orca Slicer window extends onto the left of the monitor (2x 1920x1080p monitor setup) and onto the taskbar. No such issue with the Bambu Studio slicer. But it always was (at least for a couple of months when I started using it) on the Orca Slicer.

**To Reproduce**
Maximize it and open any browser maximized on top to see the issue more clearly. Also, you will need a multi-monitor setup.

**Expected behavior**
Should not overlap with other screens and onto the taskbar when maximized

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

foreachthing, 2023-08-05:
Not happening on Win11 and Orca 1.6.3, left 1680 x 1050 and right 1920 x 1200.
Maybe try to update your graphics driver?!

DIYPERSPECTIVE, 2023-08-06:
> Not happening on Win11 and Orca 1.6.3, left 1680 x 1050 and right 1920 x 1200. Maybe try to update your graphics driver?!

It's up to date. I don't have this issue with the Bambu Studio slicer, it is clearly something in the Orca that makes this bug. I don't remember if I ever saw a bug like this ever with any program. I would just rescale the window but Orca doesn't save the window size and position on exit.

bani6809, 2023-08-06:
Windows 10, dual 4k displays, I don't get this bug with orca 1.6.3

DIYPERSPECTIVE, 2023-08-25:
Issue FIXED with v1.6.6

closed 2023-08-25 by DIYPERSPECTIVE