OrcaSlicer/OrcaSlicer #1703: No Object_height in klipper

persano, 2023-08-05:
I would like to know if someone know how to add this variable to the gcode. I have not seen it anywhere

It works with other slicers but never worked for me with orca slicer. This is not new, it happened to me since the first time I started using the slicer. It is a minor thing, but it is great to have more information regarding the files you are printing

[image]

SoftFever, 2023-08-05:
Thanks for reporting it.
Will take a look

SoftFever, 2023-08-10:
Hi @persano 
can you share your gcode?
also what's your Moonraker version? And what frontend are you using, fluidd or mainsail?

It works fine on my machien with Mainsail

[image]

persano, 2023-08-10:
@SoftFever here is the GCODE. 

[Uploading xyzCalibration_cube_PLA_22m57s.zip…]()

Sineos, 2023-08-10:
Try https://docs.mainsail.xyz/overview/slicer/prusaslicer

persano, 2023-08-10:
> Try https://docs.mainsail.xyz/overview/slicer/prusaslicer

[image]

persano, 2023-08-10:
Here is the modified file, @Sineos 

[xyzCalibration_cube_PLA_25m25s.zip](https://github.com/SoftFever/OrcaSlicer/files/12315833/xyzCalibration_cube_PLA_25m25s.zip)

Sineos, 2023-08-11:
[image]

Strange. Generally it works on my end, however your file indeed does not show any height. Sorry, no idea, but I'd suppose it is some setting thing

SoftFever, 2023-08-11:
@Sineos @persano 
I think I found something.

Moonraker rely on the custom layer change gcode to calc object height.
Probably your machine profile don't have that one or you removed it.

Put this in before layer change gcode
```
;BEFORE_LAYER_CHANGE
;[layer_z]
G92 E0
```
and this in Layer change gcode
```
;AFTER_LAYER_CHANGE
;[layer_z]
```
[image]

It should fix the issue.
Let me know what machine profile are you using, I will fix it.

persano, 2023-08-11:
> @Sineos @persano I think I found something.
>
> Moonraker rely on the custom layer change gcode to calc object height. Probably your machine profile don't have that one or you removed it.
>
> Put this in [...]

Creality Ender 3 V2, which I used as a base for my Creality CR6. Both run marlin by default but I am using klipper with my machine.

And I can confirm that with the BEFORE_LAYER_CHANGE part now I can see it!

[image]

[image]

SoftFever, 2023-08-11:
Cool, thanks for the confirmation.

closed 2023-11-10 by persano