OrcaSlicer/OrcaSlicer #1708: Orcaslicer will not preview above layer 1 in calibration test

tastyratz, 2023-08-05:
**Describe the bug**
I am running the new 1.64 beta 2 on W10 22H2. I do not know if this behavior existed before.

If I generate a PA calibration tower and slice, the preview automatically goes down to layer 1 and will not allow me to adjust the slider to show other layers.
I do not see this behavior if I slice a normal part outside of that calibration. A PA pattern test does not have this problem.

tastyratz, 2023-08-05:
Screenshot for posterity 
[image]

SoftFever, 2023-08-06:
@tastyratz 
Thanks for reporting it.
I probably should have enable the `detect thin walls`
Could you try to enable this option manually? that should fix the issue
[image]

tastyratz, 2023-08-06:
> @tastyratz Thanks for reporting it. I probably should have enable the `detect thin walls` Could you try to enable this option manually? that should fix the issue [image]

Unfortunately, the checkbox for detect thin walls is grayed out for me. It's something I've been noticing lately but from what I can read online that option typically ends up creating more weird artifacts than improving quality so I wasn't worried.
I don't see anything in a tooltip saying why or when that option might not be selectable. What would make it grayed out?

tastyratz, 2023-08-06:
Curiously, this option is also not selectable for me on my 1.63 install on my laptop however I am able to generate a successful tower without it there

[image]

Habag111, 2023-08-07:
"detect thin walls" is available only for classic wall generation, not arachne (at least that's how I understand it). turning it on fixes the PA tower for me as well (it should be enabled by default when calibrating with the PA tower imo)

SoftFever, 2023-08-07:
I will revert the changes I made in beta 2 for PA tower.
It did a poor job when line width is not around 0.45

tastyratz, 2023-08-10:
I was able to get this to generate btw by checking detect thin walls. This is not automatically checking when switching to classic mode from Arachne.

This checkbox should probably also have a tooltip text addition along the lines of "This option is not compatible with Arachne and can only be checked when using the Classic Wall Generator".

closed 2023-09-07 by SoftFever