OrcaSlicer/OrcaSlicer #1712: Cannot Select AMS Filament, List is Empty?

jucastro5, 2023-08-06:
**Describe the bug**
In the AMS section on the device tab, I cannot select what filament is loaded in the AMS. There are work arounds where I can set it on the Handy app/Bambu slicer and it syncs to Orca, but ideally I should be able to set it in Orca. 

Reinstalling or wiping /appdata/roaming/orcaslicer doesnt fix it. 

Issue also reported on the bambulab subreddit:
https://www.reddit.com/r/BambuLab/comments/15by4d3/unable_to_select_filament_type_with_ams_in_slicer/

[image]

**Expected behavior**
[image]

**Printer model**
P1S with AMS

OS: Windows 11
Version: Orca Slicer 1.6.4-beta2 (also had this issue with V1.6.3)

SurgeHawk, 2023-08-09:
Can confirm this issue. I did not have this issue when I started to use Orca. I don't know what changed (Maybe an AMS update?, I know I did not manually update AMS). I did notice this in 1.6.3 and now in 1.6.4 beta 2

kylek29, 2023-08-09:
> Can confirm this issue. I did not have this issue when I started to use Orca. I don't know what changed (Maybe an AMS update?, I know I did not manually update AMS). I did notice this in 1.6.3 and now in 1.6.4 beta 2

Bambu just released an update to this with their latest firmware (to deal with the calibration / loading prior calibrations) and the 1.7+ Studio version, I don't think that has been merged forward into OrcaSlicer yet.

Trance2289, 2023-08-10:
Can confirm this issue, as well.

slynn1324, 2023-08-14:
Same issue here - on all platforms - Mac/Windows/Linux.

slynn1324, 2023-08-15:
I did some digging, and it appears that the only issue is that the P1S has not been pulled down from BambuStudio into a configuration file.    

You can patch the current version by looking into the installation directory for the "config.json", and adding the following config to the "printers" array:

`{
            "display_name": "Bambu Lab P1S",
            "func": {
              "FUNC_CHAMBER_TEMP": false,
              "FUNC_FIRSTLAYER_INSPECT": false,
              "FUNC_AI_MONITORING": true,
              "FUNC_LIDAR_CALIBRATION": false,
              "FUNC_BUILDPLATE_MARKER_DETECT": false,
              "FUNC_FLOW_CALIBRATION": false,
              "FUNC_MONITORING": false,
              "FUNC_MEDIA_FILE": false,
              "FUNC_VIRTUAL_CAMERA": false,
              "FUNC_PRINT_WITHOUT_SD": false,
              "FUNC_ALTER_RESOLUTION": false,
              "FUNC_PRINT_ALL": false,
              "FUNC_VIRTUAL_TYAY": true,
              "FUNC_EXTRUSION_CALI": true
            },
            "camera_resolution": [ "720p" ],
            "bed_temperature_limit": 100,
            "model_id": "C12",
            "compatible_machine":["BL-P001", "BL-P002", "C11"],
            "printer_type": "C12",
            "ftp_folder" : "sdcard/",
            "printer_thumbnail_image": "printer_thumbnail_p1p"
        },`

This is a small PR - so hopefully it'll get merged up quickly.

SurgeHawk, 2023-08-16:
> I did some digging, and it appears that the only issue is that the P1S has not been pulled down from BambuStudio into a configuration file.
> 
> You can patch the current version by looking into the installation directory for the "config.json", and adding the following config to the "printers" array:
> 
> ```
> {
>             "display_name": "Bambu Lab P1S",
>             "func": {
>               "FUNC_CHAMBER_TEMP": false,
>               "FUNC_FIRSTLAYER_INSPECT": false,
>               "FUNC_AI_MONITORING": true,
>               "FUNC_LIDAR_CALIBRATION": false,
>               "FUNC_BUILDPLATE_MARKER_DETECT": false,
>               "FUNC_FLOW_CALIBRATION": false,
>               "FUNC_MONITORING": false,
>               "FUNC_MEDIA_FILE": false,
>               "FUNC_VIRTUAL_CAMERA": false,
>               "FUNC_PRINT_WITHOUT_SD": false,
>               "FUNC_ALTER_RESOLUTION": false,
>               "FUNC_PRINT_ALL": false,
>               "FUNC_VIRTUAL_TYAY": true,
>               "FUNC_EXTRUSION_CALI": true
>             },
>             "camera_resolution": [ "720p" ],
>             "bed_temperature_limit": 100,
>             "model_id": "C12",
>             "compatible_machine":["BL-P001", "BL-P002", "C11"],
>             "printer_type": "C12",
>             "ftp_folder" : "sdcard/",
>             "printer_thumbnail_image": "printer_thumbnail_p1p"
>         },
> ```
> 
> On MacOS -> /Applications/OrcaSlicer.app/Contents/Resources/config.json Windows -> <install/extract dir>/resources/config.json
> 
> This is a small PR - so hopefully it'll get merged up quickly.

This worked for me. Just had to change a couple line so that way it didn't break the other machines. 

Here is the whole config that allows all machines to still function:
`{
    "printers": [
        {
            "display_name": "Bambu Lab P1P",
            "func": {
                "FUNC_CHAMBER_TEMP": false,
                "FUNC_FIRSTLAYER_INSPECT": false,
                "FUNC_AI_MONITORING": false,
                "FUNC_BUILDPLATE_MARKER_DETECT": false,
                "FUNC_FLOW_CALIBRATION": false,
                "FUNC_MONITORING": false,
                "FUNC_MEDIA_FILE": false,
                "FUNC_REMOTE_TUNNEL": false,
                "FUNC_LOCAL_TUNNEL": true,
                "FUNC_VIRTUAL_CAMERA" : false,
                "FUNC_PRINT_WITHOUT_SD": false,
                "FUNC_ALTER_RESOLUTION": false,
                "FUNC_CHAMBER_FAN" : false,
                "FUNC_EXTRUSION_CALI": true,
                "FUNC_PRINT_ALL" : false
            },
            "camera_resolution":["720p"],
            "bed_temperature_limit": 100,
            "model_id": "C11",
            "printer_type": "C11",
            "ftp_folder" : "sdcard/",
            "printer_thumbnail_image": "printer_thumbnail_p1p"
        },
		{
            "display_name": "Bambu Lab P1S",
            "func": {
              "FUNC_CHAMBER_TEMP": false,
              "FUNC_FIRSTLAYER_INSPECT": false,
              "FUNC_AI_MONITORING": true,
              "FUNC_LIDAR_CALIBRATION": false,
              "FUNC_BUILDPLATE_MARKER_DETECT": false,
              "FUNC_FLOW_CALIBRATION": false,
              "FUNC_MONITORING": false,
              "FUNC_MEDIA_FILE": false,
              "FUNC_VIRTUAL_CAMERA": false,
              "FUNC_PRINT_WITHOUT_SD": false,
              "FUNC_ALTER_RESOLUTION": false,
              "FUNC_PRINT_ALL": false,
              "FUNC_VIRTUAL_TYAY": true,
              "FUNC_EXTRUSION_CALI": true
            },
            "camera_resolution": [ "720p" ],
            "bed_temperature_limit": 100,
            "model_id": "C12",
            "compatible_machine":["BL-P001", "BL-P002", "C11"],
            "printer_type": "C12",
            "ftp_folder" : "sdcard/",
            "printer_thumbnail_image": "printer_thumbnail_p1p"
        },
	    {
            "display_name": "Bambu Lab X1",
            "func": {
                "FUNC_VIRTUAL_TYAY" : true,
                "FUNC_EXTRUSION_CALI": false,
                "FUNC_LOCAL_TUNNEL": false
            },
            "model_id": "BL-P002",
            "camera_resolution":["720p","1080p"],
            "printer_type": "3DPrinter-X1",
            "printer_thumbnail_image": "printer_thumbnail"
        },
        {
            "display_name": "Bambu Lab X1 Carbon",
            "func": {
                "FUNC_VIRTUAL_TYAY" : true,
                "FUNC_EXTRUSION_CALI": false,
                "FUNC_LOCAL_TUNNEL": false
            },
            "model_id": "BL-P001",
            "camera_resolution":["720p","1080p"],
            "printer_type": "3DPrinter-X1-Carbon",
            "printer_thumbnail_image": "printer_thumbnail"
        }
    ]
}
`

closed 2023-08-17 by SoftFever

nirurin, 2023-08-22:
This has been closed, except the issue still exists. I am using the current version of orca.

PR #1817 (2023-08-17): Add P1S to config.json, copied from BambuStudio.  
Added P1S to config.json, copied from BambuStudio (https://github.com/bambulab/BambuStudio/blob/master/resources/config.json).  

Fixes #1712, as the printer type can now be identified and the filament type select list is populated correctly.