OrcaSlicer/OrcaSlicer #1715: Strange wall behavior

DVSVIDEO, 2023-08-06:
**Describe the bug**
Strange changes of direction/breaks in the wall (see below, bottom photo)

**3mf File for This Bug**
[ShallowWaterDispencer.zip](https://github.com/SoftFever/OrcaSlicer/files/12269602/ShallowWaterDispencer.zip)

**To Reproduce**
Normal slice procedure. 0.3mm layer height, 4 wall perimeter

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
1.6.4-beta
[image]
1.6.4-beta2
[image]

**Printer model**
Micromake C1 -modified (core XZ)
**Desktop (please complete the following information):**
 - OS: Win10

SoftFever, 2023-08-06:
This is caused by the the single wall on top.
Half of the area are qualified to apply single wall while other half doesn't ;)
I'll see if need to be fine tuned or allow user to change the threshold parameter

For now, as a workaround you can either turn off `single wall on top` for this model or change line width of inner walls to 0.6mm
[image]

DVSVIDEO, 2023-08-07:
Okay. I printed it with a single wall off. Thanks for the reply.
I really like this slicer, it's much easier to use and does things the others don't.
Keep up the good work (just don't make it as complicated with all the settings as the Cura. Only know the basics, but know them well...)

closed 2023-09-03 by DVSVIDEO