OrcaSlicer/OrcaSlicer #1720: Can't print with BL P1S in LAN mode

sdoh55, 2023-08-06:
**Describe the bug**
I am unable to send sliced gcode to P1S in LAN mode for printing. I get a warning that says the printer type selected when generating gcode is not consistent with currently selected printer, and when I click confirm to proceed anyway, I get an error titled "LAN Connection Failed (Sending print file)" that asks for IP address and access code input. Inputting the correct values does not solve the problem and the same error pops up.  

The printer information also shows up as "Unkown" under Device > Update.

**To Reproduce**
Put the BL P1S in LAN mode, slice model and send the result gcode to printer.

**Expected behavior**
Generated gcode should be sent to the P1S printer that is in LAN mode. Printer should be correctly recognized as "Bambu Lab P1S" under Device > Update. 

**Screenshots**
After slicing, clicking "Print plate", selecting the P1S, and clicking "send", the following warning pops up:
[image]

After clicking "Confirm", the progress bar loads briefly and the following window pops up:
[image]

Inputting the correct IP address and access code doesn't fix the issue and the same window pops up repeatedly.

Printer model shows as "Unknown" under Device > Update.
[image]

**What I have tried**
- Deleted OrcaSlicer folder under %appdata% and reconfigured printer
- Try to print while logged out of Bambu account
- Try to print while logged into Bambu account
- Put printer out of LAN mode and back to LAN mode

**Printer model**
Bambu Lab P1S

**Desktop (please complete the following information):**
 - OS: Windows 11
 - OrcaSlicer version: 1.6.4-beta2

SurgeHawk, 2023-08-10:
I also am having this issue with the P1S.

Windows 10

slynn1324, 2023-08-16:
Same issue here...

SurgeHawk, 2023-08-18:
In 1.6.4 Beta 3 this issue is still present. The only noticeable change from the OP is that the model is not listed as the Bambu Lab P1S in the Device>Update tab. When trying to send gcode to P1S in LAN mode still getting LAN connection failed, wrong IP or Access Code even though IP and Access Code are correct in Orca compared to printer. Device tab still connects normally.

alexawka, 2023-08-18:
Same issue here. If you look at plugins folder in appdata, you can see, that Orcas is considerably older, compared to Bambu.
It seems it could be that old plugin does not support P1S and the problem will be solved, if the plugin is updated.
If I replace plugin files in Orca with Bambu, slicer just downloads old version and replaces those.
[image]

Making files Read-Only does not help. Orca slicer still replaces them with older version.

baekho, 2023-08-19:
Same here.
If I add a device from the Devices tab without doing anything else after running the slicer for the first time, it appears to be added normally. However, when I try to print the modeling after that, I get an error message saying that I cannot connect to the printer, and I cannot connect to the equipment in the device tab after this.
(If click the Add Device button, it will appear in the list of devices on the network.)

SurgeHawk, 2023-08-24:
This has been resolved in the latest commit f7184c8 1.6.6

sdoh55, 2023-08-24:
@SurgeHawk Thanks for the update! 

Confirmed working in v1.6.6. Closing the issue.

closed 2023-08-24 by sdoh55