OrcaSlicer/OrcaSlicer #1723: Imported or user created filament profiles not showing after restart

SurgeHawk, 2023-08-07:
**Describe the bug**
When importing a filament profile or creating a new profile the profile is usable and can be selected for use. If you close OS and open it back up all user created or imported filament profiles are removed and not able to be selected.

**To Reproduce**
Import or create and save a filament profile.
Close OS
Open OS
Filament profiles is no longer selected and no longer shows in the filament profile list.

**Expected behavior**
Imported or user created profiles should be saved and stay in the filament profile drop down list

**Printer model**
Bambulabs P1S

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

kylek29, 2023-08-08:
I can confirm, I ran into this as well. I think it might be related to BambuLab syncing. 

Similar Issues Reported:
https://github.com/SoftFever/OrcaSlicer/issues/1623
https://github.com/SoftFever/OrcaSlicer/issues/1685

As an aside, I think it might be worth it to introduce a "sync debris" system, so instead of outright deleting profiles, they get moved to a .sync_debris (or whatever) folder, so a user still has them. Possibly have a program start-up function that clears profiles older than 90 (or whatever) days.

SurgeHawk, 2023-08-08:
> I can confirm, I ran into this as well. I think it might be related to BambuLab syncing. In the meantime, I turn off the sync functionality in OrcaSlicer -> Preferences.
> 
> Similar Issues Reported: #1623 #1685
> 
> As an aside, I think it might be worth it to introduce a "sync debris" system, so instead of outright deleting profiles, they get moved to a .sync_debris (or whatever) subfolder, so a user still has them. Possibly have a program start-up function that clears profiles older than 90 (or whatever) days.

Just adding here that I don't have sync with bambu servers enabled. I am signed into my bambu account though.

SurgeHawk, 2023-08-08:
Another update. 

I logged out of my bambu account and now the filaments stay. So this must be a bug with the server sync even though I don't have it enabled.

closed 2023-11-22 by github-actions[bot]