OrcaSlicer/OrcaSlicer #1725: AUX fan greyed out on latest beta

moust4ki, 2023-08-07:
**Describe the bug**
The Aux Fan power is greyed out in the filament panel on 1.6.4 beta (I have a P1P with official Aux Fan, and I can change the setting in 1.6.3)

**To Reproduce**
Edit a filament profile, go to Cooling, Aux fan is Greyed out. If you open the same profile in Bambu Studio or Orca Slicer 1.6.3 you can customize fan speed.

**Expected behavior**
Be able to customize fan speed

**Printer model**
P1P with Aux Fan

**Desktop (please complete the following information):**
 - OS: Windows 11 & Mac OS
 - Version 1.6.4 BETA

TexasTomBF, 2023-08-12:
Yep, same here on P1P

flockenoops, 2023-08-12:
For me, not only is it greyed out, but the fan doesn't spin, even though it is set to 80% in the profile.
I have to start the fan manually.

Nouyame, 2023-08-18:
Also have this issue on 1.6.4-beta3. Running a P1P, unable to change aux settings for any filament profile, and aux fan doesn't start regardless of filament profile settings (but can be turned on manually).

JJack98, 2023-08-19:
Anche io ho lo stesso problema.

buho29, 2023-08-28:
same :( use the profile of the p1s? I don't have a fan chamber

TexasTomBF, 2023-08-28:
If you go into the machine profile, you can enable the AUX fan there.
Then it's no longer greyed out under the filament section.

buho29, 2023-08-31:
Oh It has worked 
thank you very much

look for the profile in "C:\Users\name_user\AppData\Roaming\OrcaSlicer\system\BBL\machine"
change line "auxiliary_fan": "0" to "auxiliary_fan": "1",

akrtek, 2023-09-08:
attempted to make the change in the .json but still not seeing it available after shutting down OrcaSlicer and restarting

Dinchy87, 2023-10-13:
Same here on macOS. I hope it will get fixed some time.

closed 2024-01-19 by github-actions[bot]