OrcaSlicer/OrcaSlicer #1732: Print sequence ->By object clearance way too high

Pedro8005x, 2023-08-07:
**Describe the bug**
By object print sequence may useful to minimize warping during print with ASA. On my other printers, sliced with cura and same bed size there is no problem to print 4 models in one batch. On my P1P with OrcaSlicer I cannot place models to fit on plate. Looks like it draws clearance area with same distance in all direction (not too effective), and when I try to place second model, it looks it refuses placing within double-clearance area, I mean clearance to clearance is applied instead of model to model. 

**3mf File for This Bug**
[Hikoki v2 4x.zip](https://github.com/SoftFever/OrcaSlicer/files/12283584/Hikoki.v2.4x.zip)

**To Reproduce**
-Select Others tab->Print sequence->By object
-Try to place 4 models on plate, what included in attached 3mf

**Expected behavior**
-realistic clearance distance in 4 directions like in cura, model to model keepout sign (not clearance area to clearance area)
-sequence order selection will be fun

**Screenshots**
[image]
[image]

**Printer model**
P1P

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Version: (OrcaSlicer) 1.6.4-beta

xanderphillips, 2023-08-16:
There needs to be space between prints for the printhead's sides, front, and back
I do agree with @Pedro8005x that Orca slicer is way too conservative on it's spacing for clearance.
The ability to set print head & gantry size and relative place dimensions in relation to nozzle tip (using profiles?) for various printers would allow for tighter tolerances between print objects as the actual size of the print head and z-position of nozzle to gantry cross-arms would be configurable.

closed 2023-11-22 by github-actions[bot]

ilyasryvkin, 2024-04-22:
Try this if you srell need it https://www.youtube.com/watch?v=bsJgU9YMKkA&t=586s