OrcaSlicer/OrcaSlicer #1745: Error : "Line width too large" in vase mode

naeh, 2023-08-08:
When porting my vase mode profile from cura, OS can't slice the model showing this error : 
"Line width too large
Jump to [object.stl] (line width)".

Line width was set to 1.2mm
even with 1mm line width it refuses to slice.

is it possible de remove this limitation please ? i think it is the user's responsibility to set "reasonable" line width.
i have a 0.4 cht nozzle and at the right speed (pretty slow) i can print strong vases up to 1.6mm line width 

**Printer model**
Flsun SR, 0.4 nozzle

Thank you.

SoftFever, 2023-08-10:
>  i think it is the user's responsibility to set "reasonable" line width.

I agree to a certain extent. 
I will increase the limit to 5x the nozzle diameter. 
This will allow users to print in their creative ways while still providing a minimal sanity check

naeh, 2023-08-10:
great news, thank you very much :)

closed 2023-11-09 by naeh