OrcaSlicer/OrcaSlicer #1746: "One Wall on Top" causes walls to go through other parts of the print when multiple top layers. Seems to be creating two sets of inner walls

DHernan, 2023-08-08:
**Describe the bug**
This bug was introduced in 1.6.4 as 1.6.3 works fine. It seems that walls are generated funny when "One Wall on Top" is selected and can cross on the top surface despite no need for wall there. It actually might be caused by the fact it's generated two sets of inner walls but the slicer reports one of them as internal solid infill. I have 4 wall loops set and it's doing 6 on this first top layer. 

**3mf File for This Bug**
[Abducktion v6_PLA_53m0s.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12295994/Abducktion.v6_PLA_53m0s.3mf.zip)

**To Reproduce**
Not sure how to reproduce with other models 

**Expected behavior**
For walls not to cross randomly on the top surface 

**Screenshots**
Here is what it looks like when "Only one wall" is selected:
[image]

Here is what it looks like when unselected:
[image]

**Desktop (please complete the following information):**
 - OS: macOS 
 - Version 1.6.4-beta2

SoftFever, 2023-08-09:
@DHernan 
The surface is split because some parts(the text) of it is treated as top surface, which they are.
To me it's an intended behavoir. 

Could you post the screen in 1.6.3 for comparison?

Also it would helpful if the project file can be provided.

BrinksDaniel, 2023-08-09:
@SoftFever I don't understand how this would be intended so I'm hoping the project file explains clearer than me. All of the text is a flat top surface yet the inner wall is randomly running through it. 

Here is the project file (exported from 1.6.3): 
[Abducktion v6.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12297374/Abducktion.v6.3mf.zip)

Here is 1.6.3 screenshot:
[image]

It seems bambuStudio does the 6 wall thing (3 inner, 3 infill acting as walls) when using "**Classic**" wall generator but it doesn't cut through the text like 1.6.4:
[image]

Here is bambuStudio with "**Arachne**" wall generator: 
[image]

BrinksDaniel, 2023-08-09:
Good find! Any idea as to why that creates walls only 1.6.4? Even at 10%, it's still running walls through a top surface. Look at the bottom arrows/top surfaces in your second screenshot. There are random 4 walled shapes in them

SoftFever, 2023-08-10:
It's the effect of one wall on top.
You don't see this in 1.6.3 is because one wall on top for Arache only affect top-most surfaces.
In 1.6.4, it will apply for all tops.

DHernan, 2023-08-10:
You lost me again lol if it applies to all top surfaces shouldn’t it look the same as 1.6.3 but with one wall outlining the visible parts or am I misunderstanding? It would also mean cutting through the top surface instead of outlining is still a bug

SoftFever, 2023-08-10:
Hi

It's a bit hard to explain how to algorithm.
Anyway this is not the right place for such discussion.
Feel free to move to the discord server.

closed 2023-08-11 by SoftFever