OrcaSlicer/OrcaSlicer #1749: Slicer thinks overhangs are bridges

inline6power, 2023-08-09:
In the latest beta, if you have a speed set to say 100mm for bridges and also have on detect or slow down for overhangs, it will act as if the outside perimeter is a bridge. Instead of slowing down for overhangs it will speed up to what ever the bridge speed is

gbyleveldt, 2023-08-09:
I have this exact issue at the moment, I even went so far as to exhaustively explain the issue on Orca Discord.

I believe the issue is prevalent when using Classic mode as opposed to Arachne (I have not tested this myself as I don't use Arachne).

This model works like it should, although it detects overhangs as normal outer wall. But it still slows down like it should

[image]

[image]

Then, using the exact same project and settings, but replacing the model, it identifies overhangs like it should, but it uses the bridging speed on the overhangs, not the overhang speed

[image]

[image]

I've included the two projects for your reference

[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/12304554/Archive.zip)

inline6power, 2023-08-11:
Sure hope they fix this. I really only use classic walls and run my bridges as 150ms but now have to run them slow in order to slow down for overhangs. This is the only thing that's bothering me on beta 2

closed 2023-11-22 by github-actions[bot]