OrcaSlicer/OrcaSlicer #1751: Filter out gaps

inline6power, 2023-08-09:
Is there a way to filter out gaps like superslicer does? I have tried everything from detect narrow walls ect. Pic below is of the same part at the same layer height. Super does an a amazing job at this and just can’t get rid of it on certain parts[image]

kylek29, 2023-08-09:
Are we talking about the gaps on the red portion? The small holes that are present in the OrcaSlicer side?

If so, I've noticed this as well, it's related to Arachne (going back years across different slicers) -- if you switch to Classic, those go away. Did you have Arachne selected in SuperSlicer? Maybe it's doing something to fix that.

inline6power, 2023-08-09:
> Are we talking about the gaps on the red portion? The small holes that are present in the OrcaSlicer side?
>
>
>
> If so, I've noticed this as well, it's related to Arachne (going back years across [...]

I am talking about orca not utilizing gap fill as super slicer does. You can see in super the white gap fill is a continuous line. Where as on orca it stops and adds yellow internal wall. This is witu classic calls not Arachne. Both are set to the same standard

dartrax, 2023-08-15:
Having the same question!

mojotaker, 2023-10-22:
Having the same issue, its as if gap fill does not exist in Orca slicer

erlis, 2023-11-23:
You set the Gap Fill speed to zero and it disable gap fill, set a speed and then you have gap fill. I think we need the check box back! If I remember correctly this is how was in Prusa Slicer long time ago, but they added a check box...

dartrax, 2023-11-23:
The title and the first post is misleading here.
The question is not how to disable Gap fill.
The question is:

> I am talking about orca not utilizing gap fill as super slicer does. You can see in super the white gap fill is a continuous line. Where as on orca it stops and adds yellow internal wall.

[image]

> You set the Gap Fill speed to zero and it disable gap fill, set a speed and then you have gap fill. I think we need the check box back! If I remember correctly this is how was in Prusa Slicer long time ago, but they added a check box...

Gap fill speed mustn't be set to zero. Instead, use the "Filter out tiny gaps" threshold - even better than a check box ;-)

closed 2024-03-01 by github-actions[bot]