OrcaSlicer/OrcaSlicer #1762: Tree Supports generating in mid air

igiannakas, 2023-08-10:
**Describe the bug**
Tree supports generating in mid aid

**3mf File for This Bug**
3MF file can be found here: https://drive.google.com/file/d/1kAI4HKzYP2B18Y9GzfxTLOgutggixnc9/view?usp=share_link

**To Reproduce**
Slice model with Tree slim supports and the recommended settings

**Expected behavior**
No rogue extrusions happen

**Screenshots**
[image]

**Printer model**
Bambu Lab X1 carbon

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 1.6.4 Beta 2

igiannakas, 2023-08-11:
Also issue happening upstream with Bambu slicer. Fix appears to have been identified upstream: https://github.com/bambulab/BambuStudio/issues/2115

igiannakas, 2023-08-11:
Tested with latest code commits for 1.6.4-beta3. Issue appears a bit better (?) but still present.
[image]

rfield19, 2023-08-17:
j
#
'/..22222222'#
;/this has been fixed in bs 1.7.4.52 - would be amazing if it arrives in orca12222"/;no#

rfield19, 2023-08-17:
/
/
/
/

igiannakas, 2023-08-17:
I’ve created a PR for this as I’ve ported over the code change from Bambu studio as I needed this to work for a model I’m printing :)

closed 2023-08-19 by igiannakas