OrcaSlicer/OrcaSlicer #1763: Arachne w/ Support breaks Slowdown on Overhang feature

pennystockss, 2023-08-10:
**Describe the bug**
Arachne with Support, you do not get Slowdowns on Overhangs.

**3mf File for This Bug**
N/A

**To Reproduce**
Enable Supports with Arachne.

**Expected behavior**
To slowdown.

**Screenshots**
[image]
[image]
[image]

**Printer model**
P1P

**Desktop (please complete the following information):**
 - OS: Windows 11
 - 1.6.4 Beta 2

SoftFever, 2023-08-10:
Fixed in dev branch

HahBikes, 2023-08-10:
Hey @pennystockss, totally unrelated to this, but would you mind sharing your support settings?
I can't get them to generate like this with my profile. (See this [issue](https://github.com/SoftFever/OrcaSlicer/issues/1750) that I created)

closed 2023-11-19 by SoftFever