OrcaSlicer/OrcaSlicer #1764: External hole step bug

LD3D, 2023-08-10:
[image]
[image]
[image]

SoftFever, 2023-08-10:
Hi,
Can you please provide move info of the issue and the project file ?

LD3D, 2023-08-10:
I can not provide my file it is private. Im not sure what you need as your other request what move info?

LD3D, 2023-08-10:
**Describe the bug**
When there is a hole in the side of a part, any shape, the top of that hole makes a step in the outer wall inner layer by one path smaller then the wall should be. This is compounded if ther are 2 holes in the outer wall one above the other causing a stare step in and out on the inner side of the wall. The outer side of the wall is remaining as smoothe it is only stepping the wall on the inside of the part with holes going all the way through a part.

**3mf File for This Bug**
I can not attach the models as their private.

**To Reproduce**
Any part with walls make a hole through the outer wall a through hole say 3mm in a 3mm wall. Slice the part and the step at the top of the hole on the inside of the part will be present

**Expected behavior**
the step in the top of the wall should not be there. the wall has a consistant thickness but after the slicing it is stepped.

**Screenshots**
I have attached several screenshots to help show this in detail.

**Printer model**
Custom, Printer is not the concern here this error happens on all my printers from bambu to my custom printers.

**Desktop (please complete the following information):**
 -Windows 10

LD3D, 2023-08-10:
Make a part any shape put a hole through the side and slice the file it will step the inside at the top of the hole as my image shows above

closed 2023-11-22 by github-actions[bot]