OrcaSlicer/OrcaSlicer #1767: [Bug] Fan speeds incorrect based on layer time and cooling settings

psiberfunk, 2023-08-11:
**Describe the bug**
The slicer appears to have problems properly assigning fan speeds according to the rules set in the filament cooling tab.

**3mf File for This Bug **
See Attached:  
[FanSpeedIssue_OrcaSlicer_CaliCross_XY.zip](https://github.com/SoftFever/OrcaSlicer/files/12317919/FanSpeedIssue_OrcaSlicer_CaliCross_XY.zip)

**To Reproduce**
Slice the file, observe the fan speed, for example, on layer 16, which shows a fan speed on the whole layer of 53%, whereas the layer TIME on this layer is 32.8 seconds.  Accordingly, the fan speed **should be** 10%, and not 53% !

**Expected behavior**
Fan speed should respect the layer time rules in the filament profile settings instead of picking a seemingly random value for the layer.

**Screenshots**
[image]
[image]
[image]

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS:Win64
 - 1.6.4 beta 2 , but verified problem also exists in 1.6.3 release.

psiberfunk, 2023-08-11:
Addendum:  Please also see this similar but different (layer height =0.2 mm instead of 0.16 ) example where, on layer 19, despite the fan speed settings demanding a speed of 10, the slicer sets a speed of.. 80 ??     (layer time >>30 s!)

Project file attached.

[image]
[FanSpeedIssue_0.2Layerheight_OrcaSlicer_CaliCross_XY.zip](https://github.com/SoftFever/OrcaSlicer/files/12317941/FanSpeedIssue_0.2Layerheight_OrcaSlicer_CaliCross_XY.zip)

SoftFever, 2023-08-11:
I took a quick look and also compared the behavior with PrusaSlicer. 
It appears to have the same behavior. This is likely not a bug. 
When tracing the code, I found that the layer time was around 12 seconds instead of the 32 second shown in the final UI for fan speed calculation. Without delving deeper, it's difficult to make a conclusive judgment. 
It's possible that this could be the original layer time before applying slowdown. 
Considering the priority and the fact that this feature is mature, having existed and been tested by many users for years, I'm inclined not to change it.

PrusaSlicer project(you need to use open project to open it)
[ps_fan.zip](https://github.com/SoftFever/OrcaSlicer/files/12320475/ps_fan.zip)

psiberfunk, 2023-08-11:
Is the bug, instead then, that the layer time display is lying here ?  Perhaps that's actually what's wrong ?

I have another utterly bizarre example of fan speed mismatch to bring to the table where it applies it only to internal perimeters, but that seems like it may be a separate issue , and I will file it separately.

psiberfunk, 2023-08-15:
More information here: https://github.com/bambulab/BambuStudio/issues/2260

Bambu looked at the code and determined that the layer times used for fan speed vs ACTUAL layer time are different.  The fan-speed is based off a crude estimation that does not account for accelerations, for example.  Of course this can make the fan speeds based off of pretty inaccurate layer times in certain situations, particularly with the higher speed printers we're dealing with these days.

closed 2023-11-22 by github-actions[bot]

Knockout808, 2025-01-12:
i have this same issue with my P1S and it has made using orcaslicer impossible for ASA and ABS when doing multifilament prints. I have to switch to Bambuslicer for my multifilament prints as this bug is not present in it. It's so annoying becasue I love fuzzy skin and Orcaslicers newest fix to how walls are printed when using a model modifier to prevent fuzzy skin in certain areas.

psiberfunk, 2025-01-12:
@SoftFever  can this be re-opened ? it seems to be still a valid glitch.

evilDave, 2026-11-03:
I am seeing similar, though I don't think I agree with the assessment that it is the display of the layer time that is the issue. I was printing a model today that was indicating about 50 second layers, I had the min layer time set to 15 seconds, but the fan was still very much on for those layers. At something like 1/3 interpolated speed between the min and the max. And I can confirm that the layers were taking very close to 50 seconds to print.

One can easily confirm this by adding a cone primative, and trying to set the min fan threshold to whatever the middle layers of the cone take. The fan is active very much too early. This may just be an artefact of how I was testing this, but it does seem like very low values for min fan calculate correctly, eg 4 seconds seems to be about the right start point, but anything more than 15 or so seconds appears to be very far off - ie it will activate the fan at much longer layer times than that.