OrcaSlicer/OrcaSlicer #1770: Can't import config

fedonkin, 2023-08-11:
**Describe the bug**
I exported the config before reinstalling windows...After that I decided to import the config file...And the alert popped up: 
"**There are 0 configs imported**"

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
[image]

Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: Windows 10

theophile, 2023-08-12:
It would help to know what the config importer is looking for. I've encountered many .json files that won't import but work fine when placed in the appropriate subdirectory.

fedonkin, 2023-08-15:
> Было бы полезно узнать, что ищет импортер конфигурации. Я сталкивался со многими файлами .json, которые не импортируются, но нормально работают, если их поместить в соответствующий подкаталог.
Не подскажите  в какой каталог нужно кидать?

closed 2024-01-01 by github-actions[bot]