OrcaSlicer/OrcaSlicer #1773: Sliced object has holes and missing walls

Matei-alt, 2023-08-11:
**Describe the bug**
OrcaSlicer adds holes and removes walls when sliced and the issue is repaired completely when I opened a 3mf from PS and imported only the geometry ( as OrcaSlicer could not import the settings ).
After opening the 3mf all models slice well.

**3mf File for This Bug**
https://discord.com/channels/460117602945990666/461133450636951552/1139553526192803900
In the Voron discord, in #slicers_and_print_help .

**To Reproduce**
I haven't been able to reproduce the issue.

**Expected behavior**
The sliced object to be accurate to the provided 3d model.

**Screenshots**
[image]
[image]

**Printer model**
Ender 3 pro to Voron Switchwire

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Version: 21H2

closed 2023-11-22 by github-actions[bot]

Mr-Savag3, 2023-12-11:
I am having the same problem. I changed no settings from when I sliced this same model and printed it yesterday. ...ignore that it's a scary Roper for DnD. 
[image]

yachal-upson, 2024-02-29:
Have just encountered this. Can't think what I've changed to generate the issue. Same Solidworks base model, same STEP format. Similar slicer profiles. 

Hit print without looking closely enough and got some very interesting spaghetti

boyagogo, 2024-09-17:
I'm having the exact same problem files I sliced yesterday that were fine are now loosing walls.  Any of the previous posters find a solution?

Mr-Savag3, 2024-09-17:
I fixed it. There's a setting (I don't remember which one) that asks for either a % or mm. If you don't type mm, it assumes % and it destroys walls.

boyagogo, 2024-09-17:
> I fixed it. There's a setting (I don't remember which one) that asks for either a % or mm. If you don't type mm, it assumes % and it destroys walls.

THIS!  It was the Seam Gap modifier.  I had changed it from 10 to 8.  I suspected this but when I changed it back without the % I didn't notice the difference.  Thanks