OrcaSlicer/OrcaSlicer #1776: bridging fan speed wrong with classic wall generator

Luke56784, 2023-08-11:
**Describe the bug**
If I choose classic wall generator the fan speed is wrong in the bridging layer

**3mf File for This Bug**
[Duschknauf v1.zip](https://github.com/SoftFever/OrcaSlicer/files/12325211/Duschknauf.v1.zip)

**To Reproduce**
Slice the modell. Go to the bridging layer almost on top.
Choose classic oder arachne wall generator.

**Expected behavior**
both should run the fan on 100% for the complete layer

**Screenshots**
[image]

**Printer model**
Bambulab P1P

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Version 22H2

SoftFever, 2023-08-13:
@Luke56784 
I can't repro it.
Can you save it as project then share it?

[Duschknauf v1.zip](https://github.com/SoftFever/OrcaSlicer/files/12328701/Duschknauf.v1.zip)

Luke56784, 2023-08-13:
Ah it was my fault.
Now it is a project file.

[Bridging_error_project_file.zip](https://github.com/SoftFever/OrcaSlicer/files/12329293/Bridging_error_project_file.zip)

I used 1.6.3

lhndo, 2023-08-13:
similar to : https://github.com/SoftFever/OrcaSlicer/issues/1728

Luke56784, 2023-08-25:
@SoftFever 
Could you reproduce it?

closed 2023-12-01 by github-actions[bot]