OrcaSlicer/OrcaSlicer #1784: I-Fast

KnowKontrol, 2023-08-12:
Could you please please add the Qidi I-Fast to the list of printers! PLEASE PLEASE I LOVE YOU!

DVSVIDEO, 2023-08-13:
Add your own. You have the possibility to adjust the size of the bed, the Z height. Then copy the settings from the other slicer and you're done.

donnydavis, 2023-10-25:
Good luck. They use a firmware that makes you have to sneaker net files to the printer. 

In case you need the parameters for the machine, they can be found here. 

https://github.com/QIDITECH/Qidi-Print/blob/main/resources/definitions/i-fast.def.json

FrancescoFR, 2023-11-21:
how do I add this printed in orcaslicer? Thank you

closed 2024-02-27 by github-actions[bot]

mirspan, 2025-06-15:
I created a profile for QIDI i-Fast. Must test! To import into OrcaSlicer.

[Printer presets.zip](https://github.com/user-attachments/files/20744696/Printer.presets.zip)

stangs4, 2025-10-19:
I would also like to utilize Orca slicer with a Qidi i-Fast,. Can anyone please help resolve these issues? TIA

Myuavbug66, 2026-03-12:
(Printer profile a test to material print profile and to screenshots) Wrong
1 - Printer Profile
1 - Print File ( 2 Material test ) ( My actual test was ASA and TPE. I don't think PLA and TPU like each other so set up test only )
2 - Screenshots

I'm new to posting but I don't see any reason why I shouldn't share my findings if I find something that works ( I-fast It is great hardware but the software sucks ) and Qidi Has left us behind!!!