OrcaSlicer/OrcaSlicer #1787: Tree supports ignore the adaptive layer height parameter

Faasteer, 2023-08-13:
**Describe the bug**
I had a problem with tree supports not printing properly - significant underextrusion, collapses, breaks and failures. I found an old bug that mentioned flow issues with adaptive layer height for supports, which corresponds to the location that always fails for this print.

So I tried disabling this parameter and found that it doesn't work at all. No matter what style of tree supports I choose, if I turn off the adaptive layer height, the supports are still sliced with different layer heights. 

**3mf File for This Bug**
[OneWheelFender.zip](https://github.com/SoftFever/OrcaSlicer/files/12329612/OneWheelFender.zip)

**To Reproduce**
Just slice the attached file and look at the layer height view.

**Expected behavior**
A working support tree that won't ignore my settings. Or a fix so that they print correctly even with variable layer heights - I'm convinced this still doesn't work, even though this bug has been marked as resolved.

**Screenshots**
OrcaSlicer:
[image]

BambuStudio (where I used the same file):
[image]

**Printer model**
Bambu Lab P1S with AMS, but this was printed with a direct filament guide (too big spool for AMS).

**Desktop (please complete the following information):**
 - Windows 11 Pro, 22H2 (22621.2134)
 - Orca Slicer 1.6.4-beta2

luisillo26, 2023-10-14:
I'm having a similar issue with the combination os using adaptive layer heigh and automatic tree supports.
The supports don't touch the model even with "Independent support layer height" turned off:

[image]

Once I change the supports type to Normal (auto) it adjusts the support height correctly:

[image]

I'm running version 1.7.0 on MacOS 13.6.

Ocelot08, 2023-11-10:
I believe I'm having a similar bug. Using either variable layer heights has what seems to be tree supports trying to support the "block" that applies the layer variation. That has my tree supports moving above my model.

[image]
[image]

Garethgb, 2024-01-05:
Can confirm this is still happening with adaptive layers and tree support, whether its auto or manual, the support just doesn't reach the item.

[image]

closed 2024-04-13 by github-actions[bot]

jbcpollak, 2024-06-02:
This appears to still be an issue

Maxell2904, 2024-06-22:
This is still an issue