OrcaSlicer/OrcaSlicer #1788: Internal Bridge Infill does not generate properly

rendition36, 2023-08-13:
**Is your feature request related to a problem? Please describe.**
Changing the value of "Bridge Infill Direction" option does nothing, and auto internal bridge generations are always weird and sub-optimal. Like for example with the same shape one size bridge is 25 degrees and others are 90 degrees or sometimes a weird offset of like 5 degrees, they are almost generating randomly and causes top surface imperfections

**Describe the solution you'd like**
To make the function actually work, or maybe smarter internal bridge generation

**Describe alternatives you've considered**
Tried so many different settings and orientations nothing helps.

BenRoe, 2023-09-06:
I noticed the same problem.

Here some pictures to show the problem.

[image]

[image]

Atobers, 2023-09-15:
To add on to this topic, internal bridge infill does not generate properly over sparse infill, it starts to attempt to bridge over thin air instead of stretching to the nearest sparse infill. In curved like infill bridges like in my example, internal bridge support thickness option does not work as it will attempt to create a curved bridge which is impossible to replicate in thin air.

Using Orcaslicer 1.7.0
[image]

Compare how the same layer looks on Prusaslicer 2.6.1 comparable settings
[image]

mxlcxlm, 2023-09-19:
I've been experiencing this bug for a while now. I'm currently on the latest FW and it's still present. This seems kinda major...

mxlcxlm, 2023-09-20:
@Atobers I noticed that when you turn on variable layer height, it seems to generate the internal bridge corrently... Would you mind re-slicing your model to see if thats the case for you as well?

Atobers, 2023-09-20:
> @Atobers I noticed that when you turn on variable layer height, it seems to generate the internal bridge corrently... Would you mind re-slicing your model to see if thats the case for you as well?

@mxlcxlm  The pictures I have in my previous post are with variable layer height on.

To test again, I've went to the same layer for one that does not have variable height and one that does, the issue remains the same. Note the different print times, the one with the longer one is with variable height

Variable height on
[image]

Variable height off
[image]

This is what I am slicing, perhaps I am doing something wrong that I can't find: https://www.thingiverse.com/thing:3518207

lhndo, 2023-10-05:
I'd like to confirm that Infill direction setting is still broken in 1.7.0, and we could reproduce it between multiple people.

dartrax, 2023-10-10:
Another example. When Infill is aligned rectilinear at 0 degrees the result seems to be unprintable...

[image]

dartrax, 2024-01-09:
Is this still an issue? I‘ve seen an improvement but don‘t know if all is right now.

phnbk, 2024-01-10:
BambuStudio 1.8.2.56 completely ignores the top infill layer when generating the internal bridge layer leading to the bridge hanging in the air. This happens with any infill and the manual bridge direction is also ignored. This easily brakes any large flat surfaces at low infill densities. And it also makes infills that have don't have (roughly) orthogonal lines in the top layer (like gyroid) quite dangerous in such models, as the bridges can end up hanging in the air across the whole build.

In comparison PrusaSlicer ensures that the bridge actually ends on the next infill. Please do it as PrusaSlicer!

igiannakas, 2024-01-22:
> > Is this still an issue?
> 
> 1.9.0 still doesn't create Internal Bridges on sloped surfaces.
> 
> <img alt="Untitled" width="891" src="https://private-user-images.githubusercontent.com/68766/295128048-7d8e5145-ffbb-4192-a29f-f65e79932129.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5MzE2NjQsIm5iZiI6MTcwNTkzMTM2NCwicGF0aCI6Ii82ODc2Ni8yOTUxMjgwNDgtN2Q4ZTUxNDUtZmZiYi00MTkyLWEyOWYtZjY1ZTc5OTMyMTI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDEzNDkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5NzMwZDRkNWUxOTExY2YwZGRlMzFiODBiMWQ5NTUwYmVhNTUwYTg2ZTE5YWM2OWNhNTczZWYxMWM4ODFmYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yhEba9K-rUdQQDra-TbxBBgFhOqDDmAsS9zY212AFxI">

This has been fixed in the latest 2.0-dev.

https://github.com/SoftFever/OrcaSlicer/pull/3319

arough007, 2024-03-18:
The original problem of this issue is still there with the 2.0.0-beta.
Getting all kinds of wonky angles
[image]

dartrax, 2024-06-17:
I see the same with the 2.1.0-rc

However, I noticed that the tooltip states: "...for external bridges". But this is an internal bridge.
So either the correct setting for internal bridge angle is missing (which would be a feature request) or the tooltip is wrong.

slothy89, 2024-06-27:
Still persistent in 2.1.1 Stable. There appears to be no way to control the _Internal_ Infill Bridge layer direction.
The "Bridge Infill Direction" setting only affects External Bridges, as denoted in the Tool-Tip.

Daredevll, 2024-08-24:
Not sure if the proper issue to comment on, but here's how 90% of my external bridges get generated in Orca. Almost all of my bridges' "infill" are generated in mid-air and do not touch walls. Any remedy? 
[image]

igiannakas, 2024-08-24:
> Not sure if the proper issue to comment on, but here's how 90% of my external bridges get generated in Orca. Almost all of my bridges' "infill" are generated in mid-air and do not touch walls. Any remedy? ![poor bridging](https://private-user-images.githubusercontent.com/16118861/361167354-469df8f8-5148-497f-8b82-37ef80b36816.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1MjkyMTYsIm5iZiI6MTcyNDUyODkxNiwicGF0aCI6Ii8xNjExODg2MS8zNjExNjczNTQtNDY5ZGY4ZjgtNTE0OC00OTdmLThiODItMzdlZjgwYjM2ODE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI0VDE5NDgzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViMWNmMTUyYzg3MDQxMWI2MmY2ZDJkZjg1YjdjODk3N2ZhMzJlNWQwMWQ1MTdiY2U2OWFlZjMyZDMyYWI5OGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G7rKzdIr9x6ooMQHozYa6zUEz_xut3WLGPy3ozVV6Vc)

Have you tried with ensure vertical thickness to all?

andcolla80, 2024-09-04:
Same here. Internal Bridge is only generated align to Y axis.

Daredevll, 2024-09-10:
> Have you tried with ensure vertical thickness to all?

Never messed with ensure_vertical_shell_thickness and it's set to "All" by default

antidotcb, 2024-10-23:
> Not sure if the proper issue to comment on, but here's how 90% of my external bridges get generated in Orca. Almost all of my bridges' "infill" are generated in mid-air and do not touch walls. Any remedy? 

Set Internal bridge filtering to No filtering instead of Disabled.
it will generate all required internal bridges, but it will be a little bit too much.
[image]

closed 2025-01-30 by github-actions[bot]