OrcaSlicer/OrcaSlicer #1799: Request for a Creality Ender-3 V2 Neo profile

mattiasthalen, 2023-08-14:
I can see that there is a Creality Ender-3 V2 profile and I've tried modifying it for the neo, but I'm out of luck. Hence my request :)

TheBest6337, 2023-08-14:
If your neo has Klipper I could provide you my profile

mattiasthalen, 2023-08-14:
I'm afraid not, it's my first printer and I've just had it for two months.
Not even sure where to start with getting Klipper on it.

sherifsurf, 2023-08-15:
watch videos for (sonic pad) , this will help you get started with Klipper 
https://youtu.be/UtlUSF9VCOM

TheBest6337, 2023-08-15:
Please don't go with the sonic pad. It has old AF Klipper Firmware and isn't rooted.

leoaf79, 2023-11-27:
Me too, please.

lambtho12, 2023-12-05:
Why not simply copy the ender 3v2 and just change a few settings like it is done in prusa slicer?

PR #3434 (2024-01-14): New generic profile for Ender 3 V2 Neo
This PR should fix #1799, #2584 and #3345 
I don't have that printer (anymore), so I can't confirm it works? I just used Ender 3 V2 settings, changed some of them (accel, jerk, speed, size doesn't change), added thumbnail format (200x200 JPG).
It uses the basic filament profiles.
So it's up to users to test, make suggestions and improve that profile !
closed 2024-01-14 by SoftFever