OrcaSlicer/OrcaSlicer #1801: change text shape algorithm (project every point perpendicular to the corresponding surface)

discip, 2023-08-14:
## First of all, thank you for this very much appreciated work, which I think will be what we've all been waiting for, one single tool to rule them all, by combining all the features we like on different slicers.
<br>

The current behavior of the **`text shape`** is not ideal when it comes to placing text as an imprint on a curved surface, since the word wraps around e. g. a cylinder, but the individual letters do not wrap around the object.

As a result, only parts of a letter are printed on a curved surface (see attached picture).
<br>

It would be really nice if the algorithm of the **`text shape`** tool would project every point of a single letter perpendicular to the surface on which it should be placed on (not like in prusa slicer! [not perpendicular to surface on every point]).
<br>

I'm sure it's not an easy task, but I think there are plenty of skilled programmers here who could do it.
If I could, I would definitely do it.
<br>

btw:
Since I am on it:
Maybe the _**cut** feature_ could be added to the **`text shape`** menu too. So users would not need to convert every text by hand to a **`Negative Part`**.

thanks in advance
kind regards

discip, 2023-08-22:
@SoftFever
Prusa Slicer has finally implemented the right way of embossing text onto curved surfaces:
[image]

## Could we please get this too?

discip, 2023-09-10:
@Noisyfox
Good evening Sir,
I was very thrilled when PS initially announced the, at that time new, feature of being able to put text on an object directly in the slicer. But than again I found it to be lacking the option to project the text correctly onto the object, which was luckily changed with the last update.
Since you are obviously very good at porting features from PS to OrcaSlicer I dare to ask, if you are willing to port this newly reworked text feature as well?

I hope this makes sense. 😀
To be clear from the start: Of course, I am aware that this is not an easy task and that you have priorities.
But asking never hurt anyone yet, right? 😊

thanks in advance
kind regards

Noisyfox, 2023-09-11:
Might give it a shot.

discip, 2023-09-11:
Thanks for replying!

That would be really awesome!

discip, 2023-09-11:
Thanks for replying!

That would be great!

discip, 2023-10-14:
Good afternoon @Noisyfox,
I know you are really busy, but are you planing to implement this too? 👉🏻👈🏻 😅

thanks in advance
kind regards

Noisyfox, 2023-10-16:
> Good afternoon @Noisyfox, I know you are really busy, but are you planing to implement this too? 👉🏻👈🏻 😅
> 
> thanks in advance kind regards

I'm waiting for the porting of measuring tools to be completed by @SoftFever then I'll start working on this one. The measuring tools might need some major code refactoring of the GUI module which might also be useful for this one.

discip, 2023-10-16:
## Great! Then the wait is definitely worth it! 👍🏻

Please let me know as soon as you have something to test. 😁

PR #2819 (2023-12-09): Port Emboss & SVG gizmo from PrusaSlicer
This PR ported the Emboss & SVG gizmo from PrusaSlicer.

- [x]  Port Emboss gizmo
- [x]  Port SVG gizmo
- [x] Use Orca color scheme for gizmo dialog
- [ ]  ~Support reading text info from old 3mf format~ Very unlikely: https://github.com/SoftFever/OrcaSlicer/pull/2819#issuecomment-1824405783
- [ ] LOTS OF TESTING

[image]
[image]

Fix #1801
Fix #2487
Fix #2523
Fix #2616
closed 2023-12-09 by SoftFever