OrcaSlicer/OrcaSlicer #1802: no fix model trought netfab ?

KennyBeaudoin, 2023-08-15:
**Describe the bug**
there is no option to repair the model throught the netfab
error: 6 non-manifold edges

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
repair non manifold edges

**Expected behavior**
to repair the edges .,.. jesus

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

foreachthing, 2023-08-15:
But, it has a "Fix Model" function. Doesn't that work for you?
[image]

KennyBeaudoin, 2023-08-15:
[image]

KennyBeaudoin, 2023-08-15:
there is no more the option to repair it ... no i dont seee the fix model option when i right click on it .... i only see it on bambu studio not in orca slicer anymore ... i dont know why

foreachthing, 2023-08-15:
What version of OrcaSlicer are you using?
In 1.6.4b2 it's available.

KennyBeaudoin, 2023-08-15:
i tried to go back to 1.6.4b2 doesnt work... tried b3 doesnt show also

KennyBeaudoin, 2023-08-15:
if you want i can show you on discord on the voron live chat...

KennyBeaudoin, 2023-08-15:
it came back by magic ... idk why sometime the option goes away and sometimes it there ... pretty anoying and i dont understand why

closed 2023-11-23 by github-actions[bot]

weenis, 2024-01-04:
I use the "Fix Model" option quite frequently. I've been on v1.8.1 for a while, and I used the feature a few days ago. As of today, it mysteriously disappeared.

[image]

I've tried restarting. The only thing I haven't tried is reinstalling. I am running the standalone instance.

weenis, 2024-01-04:
I just launched v1.8.0 and "Fix Model" is there.

KubaMiszcz, 2025-02-19:
in 2.2.0 this option is gone again