OrcaSlicer/OrcaSlicer #1803: Orca Travels To Opposite Build Plate If Nozzle Wipe Not Enabled

Kylvanas, 2023-08-15:
**Describe the bug**
On the X1 Carbon, The slicer will cause the nozzle to travel to the opposite plate when "Wipe When Retracting" is not enabled during a multicoloured, multi plated print. This will cause the Nozzle to travel to the opposite plate (i.e. Plate 2, travels to Plate 1) and in turn, cause damage to the side of the printer itself.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

[https://drive.google.com/file/d/1vGFEIxQ_yfYKkXixrlKttxmm2ZK7MNkK/view?usp=drive_link](url)

**To Reproduce**
Steps to reproduce the behavior

Slice the same model, using dual printing and Prime Towers. Go to Print Settings next to the Printer Profile. Click Extruder. Uncheck Wipe When Retracting. Slice All. Check Travels. Click Plate 2 and observe travel lines to Plate 1.

**Expected behavior**
Travel stays maintained within the plate.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[https://imgur.com/a/cacsn9F](url)

**Printer model**
Bambu Lab X1 Carob with AMS

**Desktop (please complete the following information):**
macOS Ventura 13.4.1 (c)

karcaw, 2023-10-17:
I'm having a similar issue, but it does not change when i switch the Wipe When Retracting option is toggled.

I have four plates, and here is what is happening: https://imgur.com/a/3C0UESt

I believe that it is a problem here: 
https://github.com/SoftFever/OrcaSlicer/blob/25c9b43d62504db2f3df26f1451b940817ab8b9b/src/libslic3r/GCode.cpp#L691
which should look more like this:
https://github.com/SoftFever/OrcaSlicer/blob/25c9b43d62504db2f3df26f1451b940817ab8b9b/src/libslic3r/GCode.cpp#L423
With the extra `+ plate_origin_2d` added in there.

SoftFever, 2023-10-17:
@karcaw I can't reproduce this issue with nightly build. Are you able to repro it?

@Kylvanas could you attach a project file?

Kylvanas, 2023-10-18:
@SoftFever Upon testing, I can confirm this issue is present in 1.6.4-beta 3 and earlier versions _only_. This issue appears to have been resolved in 1.6.4 (Official Release) and later, as I can only seem to Reproduce it in 1.6.4-beta 3. I attempted to reproduce in 1.7.0, but could not.

Kylvanas, 2023-10-18:
[SoftFever_Project_File.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12984775/SoftFever_Project_File.3mf.zip)

PR #2441 (2023-10-18): add plate origin when looking for wipe tower
This may fix #1803, I don't have a clan way to build and test on windows
closed 2023-10-18 by SoftFever