OrcaSlicer/OrcaSlicer #1806: Support for Flashforge Creator Pro 2

avxmw, 2023-08-15:
**Is your feature request related to a problem? Please describe.**
After getting a Creator Pro 2 from Flashforge and being unhappy with their proprietary slicer I would like to move. I found a repository for 3rd party support to get it to work in PrusaSlicer here: https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile

Other than for reasons unknown the .AppImage of PS does not want to launch the post-processing script and I need to call it by hand the support works fine for my needs.

Seeing that adding Creator Pro 3 was mentioned in a recent discussion, I'd like to get support for the Pro 2, too so I can use one slicer for all my current needs.

**Describe the solution you'd like**
Support Flashforge Creator Pro 2 by creating/porting profiles and needed post-processing for Orca.

**Describe alternatives you've considered**
I could probably someday get this to work myself, however I am not knowledgeable about the actual workings of machine+slicer profiles so someone experienced could probably do this way quicker. At worst, I stay with PS with this machine.

**Additional context**
Provided someone walks me through adding custom profiles not in an official release I'm gladly trying to real world test.

Thanks for your considerations :)

akrtek, 2023-08-25:
This would be great. If possible the finder 3 as well.

thecivvie, 2023-11-16:
Willing to test, love to see CP2 support

Ferddy2500, 2023-11-29:
can we get this working on orca i would love to be able to use my creator 2 pro

Nightmare-Faux, 2024-01-05:
I'd love to see this printer be supported by orcaslicer. I don't like flashprint at all.

alsmithson, 2024-02-28:
There is a GitHub repository that contains a FlashForge Creator Pro 2 PrusaSlicer Profile: https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile

I have not used it as I am just researching before buying a Creator Pro 2.

A significant issue is that the Creator Pro 2 uses Sailfish firmware and requires different gcode than Marlin. People who use Cura or PrusaSlicer add a post-processor. One is included in the GitHub repository above.

closed 2024-06-06 by github-actions[bot]