OrcaSlicer/OrcaSlicer #1808: Error running JavaScript: Unsupported result type

ChaosBlades, 2023-08-15:
**Describe the bug**
When I launch OrcaSlicer I get this error. Everything seems to work fine however.

Launches OrcaSlicer
Get error 'Press OK'
Get error again 'Press OK'
Window goes black
Window opens again and I get error yet again 'Press OK'
OrcaSlicer then appears to work correctly

Lauching the AppImage from terminal mentions nothing about JavaScript and error messages provide no further details.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04 LTS x86_64 
 - Version: 1.6.3 (Also get the error or latest betas)

**Additional Information**
I do not get an error on Bambu Studio.

ChaosBlades, 2023-08-15:
Every once in a while the 2nd error I get is this instead.

[image]

m-sundman, 2023-08-22:
I also get this all the time. At least with 1.6.4, but haven't tried earlier versions.

discip, 2023-08-24:
@ChaosBlades
Since this is a duplicate of #219, please just rename the older issue and close this one.

ChaosBlades, 2023-08-24:
> @ChaosBlades
> Since this is a duplicate of #219, please just rename the older issue and close this one.

Different error. I have not gotten the error and the issue you linked for a while now. I was error-free for a while until I started getting the one in this thread which is as I said a different error and presents itself in a different way.

discip, 2023-08-24:
Sorry, my bad.
Judging by the comments in the linked issue I assumed this was the same.

Have you had success using the calibrations **`PA Line`** & **`PA Pattern`** with this error.

As I am getting the exact same error here on Manjaro Gnome with the latest release **[1.6.6](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.6)**.

And the mentioned calibrations don't work.

discip, 2023-08-24:
Thanks!

That's strange. 🤔

discip, 2023-08-24:
Thanks!

That's interesting. 🤔

RAP2022, 2023-08-26:
Hi,
I was having this same JavaScript warning since I start using OrcaSlicer (version 1.6.3 and then 1.6.4 beta 2). When using BambuLab Studio 1.7, I indeed didn't notice this error, but after checking the log, I can see the very same error.
In version 1.7, they log this error (and probably others) in to a log file instead of a dialog box.

After going through the BambuLab commits from  1.6 to 1.7, I found this two commits that do the above change:
https://github.com/bambulab/BambuStudio/commit/0699ee2da33408691e4299355e7a3a4596727c0c
https://github.com/bambulab/BambuStudio/commit/ca0847d0ac2f4c16f9d0f442bf7a8dad3767a13c

I applied to OrcaSlicer, and I don't have this dialog box anymore, but instead have this in the log file.

Hope this help you guys.

I also have another issue where OrcaStudio and BambuLab Studio crash without any core dump when enabling adaptive layer height. I don't have this error with PrusaSlicer.

karmakina, 2023-08-27:
I just installed OrcaSlicer on Linux Mint (based on Ubuntu) via appimage, it is constantly throwing this Javascript Unsupported errors. It first throws at start-up but if you close it, it just continues to come back clicking on various things in the program. Everything technically seems to work but you are constantly closing the stupid error -- if we could just log it instead of having a pop-up every 4 seconds, that would be lovely, since otherwise the program works!

Garbg, 2023-09-04:
> 

Same, constantly shows up, but dosen't cause any issues. It's extremely annoying. I'm using the appimage on opensuse tumbleweed

crtvrmn, 2023-09-05:
error shows up for me too, using debian12 and orcaslicer 1.6.6 AppImage

09:00:29 AM: Error running JavaScript: Unsupported result type
09:00:29 AM: Error running JavaScript: Unsupported result type

ChaosBlades, 2023-09-10:
Fixed or at least hidden on OrcaSlicer V1.7.0 Beta Release

closed 2023-09-10 by ChaosBlades