OrcaSlicer/OrcaSlicer #1813: Fan speed-up time "Only for Overhangs" is glitched, appears to turn off fan speedup prediction

psiberfunk, 2023-08-15:
**Describe the bug**
The purpose of the fan speed-up time prediction/compensation is to turn on the fan earlier to compensate for the fan spin up time.  However, the checkbox to do this "only for overhangs" appears to to nullify _any_ fan speedup prediction according to the preview, including for overhangs.

**3mf File for The Bug**
see file:  [FanSpeedupForOverhangsBroken.zip](https://github.com/SoftFever/OrcaSlicer/files/12345592/FanSpeedupForOverhangsBroken.zip)

**To Reproduce**
Slice included file.   Toggle the checkbox on/off in the machine settings and observe that with the checkbox ON that there is no pre-overhang fan activation in the preview.

**Expected behavior**
The "only for overhangs" box should only do the fan pre-spin for overhangs.. but checking it apparently effectively sets the speedup time to =0 for everything.

**Screenshots**
Here are comparison screenshots of the box checked and unchecked for a overhang, and proof that the slicer sees it as an overhang wall.

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 
 - Version 1.6.4 beta 3

psiberfunk, 2023-08-16:
@scottmudge , perhaps you have a comment on if this is intentional or a bug since you committed the feature ?

scottmudge, 2023-08-16:
The problem is a limitation of the algorithm I pulled from SuperSlicer.

If the overhang had more than one "line segment", it would work as intended. The algorithm is basically modifying the fan speed at different "gcode" points (starting points for a new straight-extrusion line), it's not inserting "new" gcode points at distances exactly matching the start-up or kick-start times. It's just finding the closest one existing gcode point/line-segment and adding a fan command there.

For parts that are basically just a bunch of straight lines, like this one, there are very few gcode points/line segments to tack on a fan speed command to.

The "Overhangs Only" option basically just ignores tacking on any fan speed commands (including pre-start) to anything that isn't an overhang.

For an organic shape, where there are thousands of tiny line segments, this isn't an issue. 

The algo can probably be modified to bypass the "Overhangs Only" modifier for the startup time, but this is the behavior intended by SuperSlicer.

psiberfunk, 2023-08-16:
So that helps clarify a little why this happens  , but what’s a little confusing here still is this this:

 
If checking the “overhangs only” button only allows fan tack on to overhangs .. what’s the point of a fan Pre-speedup at all ?  Won’t it never start speeding up until it hits the overhang anyways , which defeats the purpose ?

psiberfunk, 2023-08-16:
Also consider now that with the significant use of arcs these days , it’s reasonable that even more rounded and organic shapes will have at least some of these issues , as the arc is a single move at this stage of the game.

scottmudge, 2023-08-16:
Yes arc gcodes aren't supported with this approach, because it's not so simple to divide an arc up into smaller arcs, and unfortunately you can't just insert a fan speed command in the middle of another gcode command, since gcodes are executed sequentially. But arc gcodes weren't originally in SuperSlicer either. I'm not sure what the justification was for the overhang option in the original algorithm from SuperSlicer, and it could probably be improved, but I don't think they're really working on SS anymore, so it's something that will need to be improved in OrcaSlicer moving forward.

Arc gcodes are in a weird place. They were originally put in place to overcome the UART comm speed limitations of older 8-bit MCUs, but since most printers are switching to 32-bit MCUs, or even higher speed CANbus, even very high density small-length linear movements aren't really a problem anymore. And it's probably more consistent to avoid arcs in terms of consistent slicer performance printer-to-printer, as each printer's firmware is going to perform the arc interpolation differently (and potentially worse than the slicer would).

psiberfunk, 2023-08-16:
> And it's probably more consistent to avoid arcs in terms of consistent slicer performance printer-to-printer, as each printer's firmware is going to perform the arc interpolation differently (and potentially worse than the slicer would).

It turns out bambu’s MC does really badly if you don’t use arcs in many situations unfortunately.

More generally … It feels like this may not be possible to properly without sub-segmentation of the lines ? 

Additionally , I would point out that due to the other bug I logged #2260 , it looks like layer time estimates can be way off at this stage (vs the more accurate preview) … because apparently acceleration is not taken into account when calculating fan speeds according to @SoftFever (see #2260).

this means , I’m guessing , that any time estimates here are probably going to be off as well ?

psiberfunk, 2023-08-19:
@scottmudge & @SoftFever  To clarify.. it turns out that as long as the "only on overhangs" box is checked.. you can change the value in the fan speed-up box to anything (say 100) .. and it will have zero effect, even on organic models with complex overhangs.  I think the checkbox fundamentally disables the feature.. so i'm not sure of what it's purpose is ? 

Here's a perfect example with the fan speed-up time set to 0 and 10 seconds, respectively... look how the slice is the same in the fan-speed view:

[image]
[image]

Here's the project file, if you want to validate this:

[OverhangsNoFanBehaviorChange.zip](https://github.com/SoftFever/OrcaSlicer/files/12386342/OverhangsNoFanBehaviorChange.zip)

psiberfunk, 2023-08-19:
And before someone says "turn off arc welding" .. I tried that.. and it the behavior is identical:

[image]
[image]

closed 2023-12-01 by github-actions[bot]