OrcaSlicer/OrcaSlicer #1820: Adding Resonance Compensation to Calibration tab for easy Tuning

Lenrul, 2023-08-15:
trying to print fast is nice but with the ringing from the resonance getting from **[insert any printer]**, obviously, the print quality is going to tank.

It would be nice to add another option under the Calibration tab for Resonance Compensation that would automatically print the ringing tower for any printer to tune the frequencies. its under this [Klipper Resonance Compensation Page](https://www.klipper3d.org/Resonance_Compensation.html)

the only other alternative is using the accelerometer to get calibration but not all people would put it on there for that specific purpose hence why I requested for one.

closed 2023-11-23 by github-actions[bot]

MrIsak, 2024-01-08:
This "Easy Tuning" job would be very nice for beginners, since it's not easy to follow the guide on klipper, and find and adjust all settings in Orca.

Litto987, 2024-11-07:
Agree on the above, it's not the easiest topic for beginners and having it as part of the calibration menu would be a nice addition.