OrcaSlicer/OrcaSlicer #1825: Unable to Modify Filament Type, Color, or K-value

Dunngeon, 2023-08-16:
**Describe the bug**
In the device tab, when clicking on Ext Spool's edit button (No AMS), the window that pops up does not have the Filament, Color, or Nozzle Temperature propagated. There are no filaments in the dropdown either. As a result, I cannot adjust the K-value within OrcaSlicer because it requires those fields to be filled.

The functionality works fine in Bambu Studio, and the adjusted k-value appears to propagate over to OrcaSlicer via sync once it's saved in BS. The filament type (bambulabs petg basic) was originally set via the Bambu Handy app.

**3mf File for This Bug**
Not relevant

**To Reproduce**
Open Orcaslicer, navigate to device tab, attempt to change Filament, Color, or Nozzle temperature of an externally attached

**Expected behavior**
Propagate Filament, Color, nozzle temperature and K-value from sync, then allow me to adjust any of them, and save.

**Screenshots**
Orca
[image]
Bambu Studio
[image]

**Printer model**
P1S

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version Orcaslicer V1.6.4-beta3

SurgeHawk, 2023-08-16:
This resolved my issues. #1712

Dunngeon, 2023-08-16:
> This resolved my issues. #1712

#1712 did not resolve my issues. I tried the following:

1.  I tried copying the suggested code from #1712 into the Orca config.json with the P1S code at the top of the file.
2. Deleted all other entries for p1p, x1, and x1c
3. Directly cloning the bambu studio config.json into the directory

The issue still persists.

Additionally, I didn't note this before but the camera plugin also is inoperative for me on OrcaSlicer.

Dunngeon, 2023-08-17:
> This resolved my issues. #1712

Thanks! 

The code in 1712 ended up resolving the issue - however note that OrcaSlicer did not register the change for a few hours and multiple restarts later. 

I tested on a separate laptop and it immediately registers if you update the config and then restart your computer. Note that this fixes filament selection, color, k-value updating AND camera functionality.

Closing this issue since 1712 has a pull request open that will solve the problem.

Full Code from my config.json for anyone looking for a drop in replacement
`{
    "printers": [
        {
            "display_name": "Bambu Lab P1S",
            "func": {
              "FUNC_CHAMBER_TEMP": false,
              "FUNC_FIRSTLAYER_INSPECT": false,
              "FUNC_AI_MONITORING": true,
              "FUNC_LIDAR_CALIBRATION": false,
              "FUNC_BUILDPLATE_MARKER_DETECT": false,
              "FUNC_FLOW_CALIBRATION": false,
              "FUNC_MONITORING": false,
              "FUNC_MEDIA_FILE": false,
              "FUNC_VIRTUAL_CAMERA": false,
              "FUNC_PRINT_WITHOUT_SD": false,
              "FUNC_ALTER_RESOLUTION": false,
              "FUNC_PRINT_ALL": false,
              "FUNC_VIRTUAL_TYAY": true,
              "FUNC_EXTRUSION_CALI": true
            },
            "camera_resolution": [ "720p" ],
            "bed_temperature_limit": 100,
            "model_id": "C12",
            "compatible_machine":["BL-P001", "BL-P002", "C11"],
            "printer_type": "C12",
            "ftp_folder" : "sdcard/",
            "printer_thumbnail_image": "printer_thumbnail_p1p"
        },
		  {
            "display_name": "Bambu Lab P1P",
            "func": {
                "FUNC_CHAMBER_TEMP": false,
                "FUNC_FIRSTLAYER_INSPECT": false,
                "FUNC_AI_MONITORING": false,
                "FUNC_BUILDPLATE_MARKER_DETECT": false,
                "FUNC_FLOW_CALIBRATION": false,
                "FUNC_MONITORING": false,
                "FUNC_MEDIA_FILE": false,
                "FUNC_REMOTE_TUNNEL": false,
                "FUNC_LOCAL_TUNNEL": true,
                "FUNC_VIRTUAL_CAMERA" : false,
                "FUNC_PRINT_WITHOUT_SD": false,
                "FUNC_ALTER_RESOLUTION": false,
                "FUNC_CHAMBER_FAN" : false,
                "FUNC_EXTRUSION_CALI": true,
                "FUNC_PRINT_ALL" : false
            },
            "camera_resolution":["720p"],
            "bed_temperature_limit": 100,
            "model_id": "C11",
            "printer_type": "C11",
            "ftp_folder" : "sdcard/",
            "printer_thumbnail_image": "printer_thumbnail_p1p"
        },
	    {
            "display_name": "Bambu Lab X1",
            "func": {
                "FUNC_VIRTUAL_TYAY" : true,
                "FUNC_EXTRUSION_CALI": false,
                "FUNC_LOCAL_TUNNEL": false
            },
            "model_id": "BL-P002",
            "camera_resolution":["720p","1080p"],
            "printer_type": "3DPrinter-X1",
            "printer_thumbnail_image": "printer_thumbnail"
        },
        {
            "display_name": "Bambu Lab X1 Carbon",
            "func": {
                "FUNC_VIRTUAL_TYAY" : true,
                "FUNC_EXTRUSION_CALI": false,
                "FUNC_LOCAL_TUNNEL": false
            },
            "model_id": "BL-P001",
            "camera_resolution":["720p","1080p"],
            "printer_type": "3DPrinter-X1-Carbon",
            "printer_thumbnail_image": "printer_thumbnail"
        }
    ]
}`

closed 2023-08-17 by Dunngeon