OrcaSlicer/OrcaSlicer #1827: Disabling prime tower when printing with multiple filaments sometimes causes pretty long prints

mplonski, 2023-08-17:
**Describe the bug**
Sometimes, when printing multi-colour model without prime tower, calculated printing time is unrealistically long. So far I thought that it was purely random since it happened to me from time to time and restarting Orca (and opening my file again) usually solved the issue. When trying to reproduce it I found out that it may be related to having and not having prime tower.

When printing with this bug "enabled" and generated into gcode, print is actually taking a long time (it kinda hangs after switching colour and doesn't move at all, or does it very slowly).

**3mf File for This Bug**
Including three files: 3mf with simple project, gcode generated with prime tower enabled (Cube_PLA_25m42s.gcode) and gcode generated without prime tower enabled.

[test models.zip](https://github.com/SoftFever/OrcaSlicer/files/12364306/test.models.zip)

**To Reproduce**
 * Open CuraSlicer (tested also on 1.6.4-beta3)
 * add any model (or cube from Orca)
 * add another filament
 * add change to another filament on any layer
 * slice it, should be ok
 * disable prime tower
 * slice it, time may be unrealistically long

If it doesn't work as described above, restart Orca and try again :)

**Expected behavior**
Calculated gcode should not depend on how many times Orca was restarted, it should always be proper.

**Screenshots**
Attached: one with prime tower enabled, one without.

[image]

[image]

**Printer model**
Bambu Lab P1S (happened also on P1P)

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 13.5

closed 2023-11-23 by github-actions[bot]

christophervankammen, 2024-01-13:
I know this was closed by the bot but did it actually get fixed?

mplonski, 2024-01-13:
@christophervankammen last time I checked it wasn't, but haven't tried printing for a month.