OrcaSlicer/OrcaSlicer #1830: Camera problems in OrcaSlicer.

serpaw7, 2023-08-17:
Hi guys, I ran into a problem, turned on LAN mode on the Bambu X1 printer and the camera stopped working in Orcaslicer, when you click on playback, the Initialization Faled error appears (Not supported with LAN-only mode). I tried to do the same on Bambu studio also in LAN mode, everything works fine there, the camera shows. What could be the problem?? I also noticed that Orcaslicer slows down very much if there is no Internet, you can say it just freezes and it is not possible to work, what could this be related to??

SoftFever, 2023-08-19:
I will see what can I do for the camera.
It's a bit tricky and undocumented because it's their closed sourced plugin

I've never seen the slowdown issue. have you tried restart the application?

serpaw7, 2023-08-19:
Thank you very much if you can figure out the camera, otherwise it's not convenient to run two applications in LAN mode, make settings in one, and watch through the other)) As for the brakes, the connection is like this, if the network settings are auto on the desktop PC and the Internet is turned on, then everything works smartly, if the network settings are auto and the Internet is turned off, then OrcaSlicer slows down a lot, Bambu Studio by the way, too. If the IP address of the machine, the network mask, and the IP of the main gateway are specified in the PC settings in the network connection, then everything works fine, even if the Internet is turned off. I tried it several times.

serpaw7, 2023-08-25:
I tried adding this to the registry. 

HKEY_CLASSES_ROOT\bambu
    Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}

It did not give any result.
I checked through regedit that this entry exists.

closed 2023-10-01 by SoftFever