OrcaSlicer/OrcaSlicer #1832: Pressure advance problem with adaptive bed mesh

Mharasim1, 2023-08-17:
**Describe the bug**
Adaptive bed mesh probes only small square instead of the whole pattern

**To Reproduce**
Generate pressure advance callibration using pattern method

**Expected behavior**
Probe the whole area

TheBest6337, 2023-08-17:
Which Orca slicer version are you using?

Mharasim1, 2023-08-17:
@TheBest6337 1.6.4 beta 2

SoftFever, 2023-08-19:
This is a known issue with KAMP
I will fix it in the official release.

Aasikki, 2024-06-11:
This is closed as completed but the issue still persists in the latest version of orca

Erosion139, 2025-01-23:
> This is closed as completed but the issue still persists in the latest version of orca

bumping this because I had the same issue, though mine isnt even using adaptive at all its scanning the whole bed.