OrcaSlicer/OrcaSlicer #1834: Move out of range at the end of print

bonjorno7, 2023-08-17:
**Describe the bug**
When the print is finished it tries to move the print head up and out of the way, but it picks a location far outside the build plate.
I think this only happens in mult-plate projects for me.

**3mf File for This Bug**
Had to zip the 3mf file because GitHub didn't recognize the extension: [Hero Me.zip](https://github.com/SoftFever/OrcaSlicer/files/12373851/Hero.Me.zip)

**To Reproduce**
Create a project file with at least 3 plates; they need to be in a grid instead of just in a single row.
Slice one of the plates on the second row and show travel moves.

**Expected behavior**
It should keep the print head within the bounds of the plate.

**Screenshots**
It tries to move to a spot on the first row of plates, instead of staying within its own plate.
[image]
Klipper just cancels the print, which is fine; it's easy to clip off the bit of ooze left by the nozzle sitting at it's final position.
[image]

**Printer model**
Ender-3 Pro running Klipper from a laptop.

**Desktop (please complete the following information):**
 - OS: OrcaSlicer is running on Windows 10, Klipper is running on Linux.
 - Version: Windows 10.0.19045, Klipper idk but I updated yesterday.

** Notes **
Since it does it after the print is finished, this is not very high priority (for me at least).
Unless there's firmware out there that breaks your printer when you ask it to go out of bounds I guess.
Still, would be nice if it worked properly, and I imagine it won't be too hard to fix.

SoftFever, 2023-08-19:
@bonjorno7 
This is caused by your end print custom g-code
Please check your printable height setting in the slicer and Klipper, they should match.
Also double check your machine end gcode.

bonjorno7, 2023-08-19:
It's not the height that's the problem, it's that it goes way off the bed on the Y axis.
The first row of plates behave normally, it's the rows after that try to move to a position on the first row instead of staying within their own row.

bonjorno7, 2023-08-19:
It's not the height that's the problem, it's that it goes way off the bed on the Y axis.
The first row of plates behave normally, it's the rows after that try to move to a position on the first row instead of staying within their own row.
You can see it going to Y=409.600 in the Klipper console screenshot.

SoftFever, 2023-08-19:
> It's not the height that's the problem, it's that it goes way off the bed on the Y axis.
> The first row of plates behave normally, it's the rows after that try to move to a position on the first row instead of staying within their own row.
> You can see it going to Y=409.600 in the Klipper console screenshot.
> I'll check my end g-code when I get on the computer tomorrow.

Ah, overlooked this.

Yes, it's a bug.
Will fix it soon

SoftFever, 2023-08-19:
Fixed in dev branch

bonjorno7, 2023-08-19:
You're quick!
Thanks for making my favorite slicer 👍

closed 2023-09-07 by SoftFever