OrcaSlicer/OrcaSlicer #1838: Mouse Ears clearance error

kolchagov, 2023-08-18:
**Describe the bug**
When using mouse ears brim with elephant foot compensation, the brim is not sticking to the object

**3mf File for This Bug**
not needed

**To Reproduce**
Enable elephant foot compensation for example 0.15 and enable "mouse ears" brim, set brim object gap to 0 (which have to be default instead 0.1 by the way)

**Expected behavior**
mouse ears brim should touch the print object

**Screenshots**
Elephant foot compensation on: [https://ibb.co/7SqLmxM](url) 
compensation off: [https://ibb.co/z2RL0kx](url)

**Printer model**
any

**Desktop (please complete the following information):**
 - OS: MacOS Monterey
 - Version [e.g. 22]

Noisyfox, 2023-08-18:
That's exactly how "Elephant foot compensation" works.

SoftFever, 2023-08-19:
@kolchagov 
Let me elaborate further:
When we set the elephant foot compensation to 0.15, the slicer will shrink or offset the first layer (of the object) by 0.15mm. This compensates for the extra expansion caused by an imperfect first layer's z-offset.

Brims, and their associated brim gaps, are generated based on the original object shape. Altering this could disrupt the actual print. Picture the red lines as the object boundaries after printing.
[image]

kolchagov, 2023-08-19:
Ok, I understand that, but still the clearance is too big. Even setting elephant compensation to 0 separates the mouse ears from the object. Classic brim works fine though.
here is a real photo: https://ibb.co/3ND6MsV

SoftFever, 2023-08-20:
@kolchagov 

From the generated gcode, everything works as expected; it might be an inconsistent first layer issue.
Which is a more suitable topic for the community forum/discord.

closed 2023-12-01 by github-actions[bot]