OrcaSlicer/OrcaSlicer #1841: Not able to launch Orca Slicer on Fedora Linux 38

magnum61, 2023-08-18:
**Describe the bug**
The app image cannot be started. Error message regarding libwebkit2gtk, but this is installed properly on the system.
System is installed from scratch - Fedora 38 KDE Spin

**To Reproduce**
Download and launch the app image 

**Expected behavior**
Launch the application

**Printer model**
Creality K1 Max, but it should not matter as the app is not even starting

**Desktop (please complete the following information):**
OS: Fedora Linux 38
Desktop: KDE Plasma

magnum61, 2023-08-18:
Meanwhile i got it working after additionally install the devel version of libwebkit2gtk.

However the Java errors persist. This wouldn't be a problem, but the import of a machine profile (Creality K1) is not persistent. The file is stored in the profile folder, but won't show up in Orca Slicer as new config.

KenricYoung, 2023-09-15:
Confirmed the launching issue. 
Helped track this issue for a user on the fedora discord for the latest 1.7.0 appimage.
For Workstation users and the Plasma spin, as @magnum61 mentioned, fedora users must install the webkit2gtk4.0 to launch the appimage. 

I used toolbx using the fedora-toolbox:38 container to run the flatpak. The container image needed the following packages to provide the specified libraries to launch the appimage.
1. fuse-libs --> libfuse.so.2
2. fontconfig --> libfontconfig.so.1
3. libglvnd-glx -->libGL.so.1
4. libglvnd-egl --> libEGL.so.1
5. gtk3 --> libgtk-3.so.0
6. gstreamer1-plugins-base --> libgstvideo-1.0.so.0
7. webkit2gtk4.0 --> libwebkit2gtk-4.0.so.37

kinghat, 2023-12-15:
not stale and relevant on fedora 39 still.

kevinpita, 2024-01-02:
Installing `webkit2gtk4.0` fixed this issue for me on Fedora 39
```
sudo dnf install webkit2gtk4.0
```

kinghat, 2024-01-02:
> Installing `webkit2gtk4.0` fixed this issue for me on Fedora 39
> 
> ```
> sudo dnf install webkit2gtk4.0
> ```

i actually had to run: `sudo dnf install '/usr/lib64/libwebkit2gtk-4.0.so.37'`

XrayIndia4, 2024-01-16:
I am on Fedora 39 (gnome) and Orca Version 1.9 (older versions have this issue too for me)

I couldn't launch either BambuStudio or OrcaSlicer, but webkit2gtk4.0-devel did work for me. I already had the non-devel version installed, but that didn't help.  I also tried BambuStudio Flatpak that recently came out and everything works there from the get go, but I'd much rather us OrcaSlicer...

Here's a link to the fedora package: [https://packages.fedoraproject.org/pkgs/webkitgtk/webkit2gtk4.0-devel/](url)

lucagervasi, 2024-01-19:
Confirm unable to have it working on Fedora 39.i'm pretty sure there is a webpage under the setup wizard as the cursor seems to "see" selectable text in some points.

[image]

bryanweatherly, 2024-02-17:
> Installing `webkit2gtk4.0` fixed this issue for me on Fedora 39
> 
> ```
> sudo dnf install webkit2gtk4.0
> ```

This also worked to fix the issue on Fedora Silverblue 39

```
rpm-ostree install webkit2gtk4.0
```

lucagervasi, 2024-02-19:
> > Installing `webkit2gtk4.0` fixed this issue for me on Fedora 39
> > ```
> > sudo dnf install webkit2gtk4.0
> > ```

Still unrendered windows.
```
Last metadata expiration check: 0:48:02 ago on lun 19 feb 2024, 10:38:36.
Package webkit2gtk4.0-2.42.5-1.fc39.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
```

> 
> This also worked to fix the issue on Fedora Silverblue 39
> 
> ```
> rpm-ostree install webkit2gtk4.0
> ```

Mine is not ostree-based :(

xPakrikx, 2024-02-24:
> Still unrendered windows.
>
> ```
> Last metadata expiration check: 0:48:02 ago on lun 19 feb 2024, 10:38:36.
> Package webkit2gtk4.0-2.42.5-1.fc39.x86_64 is already installed.
> Dependencies resolved. [...]

Same on my fedora 39.

cadriel, 2024-02-29:
I had the same issue. I installed the `webkit` dependency which allowed me to open the app. However, I then installed the bambu network plugin - and signed in, and now the app consistently crashes on startup.

vprasad1973, 2024-03-10:
FYI to any of the Fedora or other distro folks having difficulty with launching Orca app image, go ahead and try last night's Jan 22 v2.0.0 dev image from the nightly builds-- WORKS FANTASTIC!!!

magnum61, 2024-03-10:
>  go ahead and try last night's Jan 22 v2.0.0 dev image from the nightly builds-- WORKS FANTASTIC!!!

Nope, not on my device with Fedora 39. Still the webkit error. i can install it manually, but the AppImage requires it.

[image]

cadriel, 2024-03-10:
Its now working for me. The network plugin re-installed when I opened it, and now it doesn't crash - and I can see the live feed of my printer.

I did already have the webkit dependency installed, so I'm not sure if that's still a requirement - but it otherwise works flawlessy so far.

Great work!

JohnGalt1717, 2024-03-17:
I'm still getting a blank screen on the setup wizard.

dslauter, 2024-03-19:
Installing webkit fixed the OrcaSlicer on OpenSuse Tumbleweed

`sudo zypper in libwebkit2gtk-4_0-37`

Thank you!

advlogic, 2024-04-10:
Same issue, tried libwebkit and also the latest beta AppImages.  Blank setup screen on Fedora 39 KDE.

xPakrikx, 2024-04-13:
> Confirm unable to have it working on Fedora 39.i'm pretty sure there is a webpage under the setup wizard as the cursor seems to "see" selectable text in some points.
> 
> ![Screenshot from 2024-01-19 14-18-45](https://private-user-images.githubusercontent.com/2998405/298079932-070c5eb3-41f6-400f-b9a1-31ba51c6fce7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwNDY0MjYsIm5iZiI6MTcxMzA0NjEyNiwicGF0aCI6Ii8yOTk4NDA1LzI5ODA3OTkzMi0wNzBjNWViMy00MWY2LTQwMGYtYjlhMS0zMWJhNTFjNmZjZTcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTNUMjIwODQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODE4ZmY3Zjg5ZDEyMDIwNDBkMTMzNTQ5ZTJlODQ2NzI4MGVhYzZmNzNiNjU1NjE4ZWQ1MGZhMWFjZGMxN2ZkOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.NORnFLZ3Zjx0YToJuFE8wxvGADETYslq7XDWv6I2ZKk)

This was solved by this for me: https://github.com/SoftFever/OrcaSlicer/issues/4935#issuecomment-2050100610

Luke56784, 2024-04-14:
Same issue on Fedora 39 KDE.

```
sudo dnf install webkit2gtk4.0.x86_64

sudo dnf install webkit2gtk4.0-devel.x86_64
```

fixed it. It's starting now.

advlogic, 2024-04-16:
[image]

No dice

Luke56784, 2024-04-16:
> Same issue on Fedora 39 KDE in combination with the V2.0.0 release.
> 
> ```
> sudo dnf install webkit2gtk4.0.x86_64
> 
> sudo dnf install webkit2gtk4.0-devel.x86_64
> ```
> 
> fixed it. It's starting now.

I don't know why but today I could not start the app image again...

breathless19, 2024-04-22:
Just a heads up guys, the latest nightly appimage (2.1.0) works without monkeying with webkit stuff.  

I'm using Bazzite (Fedora Universal Blue) and have a Nvidia GTX 1080

advlogic, 2024-04-30:
2.1.0 makes it past the setup screen now, but then crashes after using it for 5 minutes.  Not useable at all.

hidara2000, 2024-05-03:
I can confirm that the following worked for me on Fedora 40
sudo dnf install webkit2gtk4.0 gtk3

I had some issues with dark mode either not working or the GCODE text was invisibe. My working fix at this [Github link](https://github.com/SoftFever/OrcaSlicer/issues/883#issuecomment-2092331449)

jamincollins, 2024-05-05:
> No dice. Executing export WEBKIT_DISABLE_COMPOSITING_MODE=1 does work though. I think I will come back when there is a tested Fedora package available. Seems pretty hacky to do Appimages and all these weird workarounds.

Does your system have an Nvidia GPU?

jamincollins, 2024-05-05:
Assuming your system does have a discrete Nvidia GPU, this PR should fix it:
https://github.com/SoftFever/OrcaSlicer/pull/5276

DaveSpe, 2024-07-16:
It does not fix it, still an issue in Fedora 40.

QuietWindUponTheMoor, 2024-07-17:
I'm having the exact same issue... *sigh* I thought it was just me, guess not. Been trying to fix this for a week with no luck. Using Fedora 40 with KDE Plasma 6. I've tried basically all of the suggestions mentioned here to no avail so far.

This really sucks because I have some projects to work on

corincorvus, 2024-07-22:
Same Problem like @QuietWindUponTheMoor. 2.1.1 not work on Fedora 39 and 40.

My Error:
```
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
Failed to connect to bus: Kein Medium gefunden
Failed to connect to bus: Kein Medium gefunden
[2024-07-22 12:07:16.745672] [0x00007f4e1c850680] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Authorization required, but no authorization protocol specified

(orca-slicer:13962): GLib-GObject-CRITICAL **: 12:07:16.896: invalid (NULL) pointer instance

(orca-slicer:13962): GLib-GObject-CRITICAL **: 12:07:16.896: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

DaveSpe, 2024-07-22:
> I'm having the exact same issue... _sigh_ I thought it was just me, guess not. Been trying to fix this for a week with no luck. Using Fedora 40 with KDE Plasma 6. I've tried basically all of the [...]

You can launch the software from console to get through the wizard with this command...

```
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ~/Applications/OrcaSlicer_Linux_V2.1.1_f96f5666c033dc24e3f0b0f3c57a4402.AppImage
```

corincorvus, 2024-07-23:
> 
> You can launch the software from console to get through the wizard with this command...
> 
> ```
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ~/Applications/OrcaSlicer_Linux_V2.1.1_f96f5666c033dc24e3f0b0f3c57a4402.AppImage
> ```
> 
> Once you do that once and go through the Wizard, subsequent launches you will be able to start the software normally.

If i use your Command:
```
(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.273: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.274: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.280: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.280: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.286: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:19.416: gtk_widget_set_size_request: assertion 'width >= -1' failed
Failed to get GBM device
Failed to get GBM device
Datei oder Verzeichnis nicht gefunden

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.382: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.382: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.382: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.382: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.384: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.384: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.385: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.385: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.386: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.386: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.387: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.387: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7050): GLib-GObject-CRITICAL **: 09:04:20.387: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7050): Gtk-CRITICAL **: 09:04:20.387: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:7050): CRITICAL **: 09:04:20.391: Cannot register URI scheme wxfs more than once

** (orca-slicer:7050): CRITICAL **: 09:04:20.391: Cannot register URI scheme memory more than once
Failed to get GBM device
Failed to get GBM device
Failed to get GBM device
``` 
[image]

FinnKrass, 2024-07-26:
> You can launch the software from console to get through the wizard with this command...
> ```
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ~/Applications/OrcaSlicer_Linux_V2.1.1_f96f5666c033dc24e3f0b0f3c57a4402.AppImage
> ```

This works, thanks. Launches after going through the wizard still blanked though, so just making a script to launch this command may be the best solution for now :)

corincorvus, 2024-07-30:
Today, after some Updates, the Error changes:

Command:
```
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ~/Programmdaten/Apps/OrcaSlicer_Linux_V2.1.1_f96f5666c033dc24e3f0b0f3c57a4402.appimage

```

```
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
[2024-07-30 15:00:43.129688] [0x00007f14bc664680] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Authorization required, but no authorization protocol specified

(orca-slicer:29580): GLib-GObject-CRITICAL **: 15:00:43.267: invalid (NULL) pointer instance

(orca-slicer:29580): GLib-GObject-CRITICAL **: 15:00:43.267: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

dumb-taco, 2024-08-05:
I had the same problem but with the installation 'sudo dnf install webkit2gtk4.0 gtk3' I was unexpectedly able to start and use Orcaslicer 2.2.0 with [Gear Lever](https://github.com/mijorus/gearlever) without any problems.

NoSadBeHappy, 2024-08-07:
> > Installing `webkit2gtk4.0` fixed this issue for me on Fedora 39
> > ```
> > sudo dnf install webkit2gtk4.0
> > ```
> 
> i actually had to run: `sudo dnf install '/usr/lib64/libwebkit2gtk-4.0.so.37'` on f39 kde

This worked for me on fedora 40, thanks!

witc-git, 2024-10-26:
Installing webkit2gtk4.0 and launch with WEBKIT_DISABLE_COMPOSITING_MODE=1 variable gets the startup wizard working, but only the menus work, the render & preview panes are blank.

Fedora 40
Gnome 46 (primarily Wayland)
nVidia
V2.1.1 Appimage with Gear Lever

Aleblazer, 2024-12-26:
> Installing webkit2gtk4.0 and launch with WEBKIT_DISABLE_COMPOSITING_MODE=1 variable gets the startup wizard working, but only the menus work, the render & preview panes are blank.
> 
> Fedora 40 Gnome 46 (primarily Wayland) nVidia V2.1.1 Appimage with Gear Lever

Having the same issue on Bazzite 41, KDE 6 Wayland, nVidia, v2.2.0 Appimage with Gear Lever. Menus work, but nothing 3D is rendering in the preview pane.

dslauter, 2024-12-26:
> Having the same issue on Bazzite 41, KDE 6 Wayland, nVidia, v2.2.0 Appimage with Gear Lever. Menus work, but nothing 3D is rendering in the preview pane.

I used this flatpak without issue. https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-2

Aleblazer, 2024-12-26:
> I used this flatpak without issue. https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-2

Just installed this flatpak, having the same issue, menus render, but the preview pane is not rendering.
[image]

2nOrderEDO, 2024-12-27:
> > > > Installing webkit2gtk4.0 and launch with WEBKIT_DISABLE_COMPOSITING_MODE=1 variable gets the startup wizard working, but only the menus work, the render & preview panes are blank.
> > > > Fedora 40 Gnome 46 (primarily Wayland) nVidia V2.1.1 Appimage with Gear Lever
> > > 
> > > 
> > > Having the same issue on Bazzite 41, KDE 6 Wayland, nVidia, v2.2.0 Appimage with Gear Lever. Menus work, but nothing 3D is rendering in the preview pane.
> > 
> > 
> > I used this flatpak without issue. https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-2
> 
> Just installed this flatpak, having the same issue, menus render, but the preview pane is not rendering. ![OrcaSlicerRenderIssue](https://private-user-images.githubusercontent.com/60912320/398799386-cbf0ad40-ba97-4443-8e87-2072214b1fd6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyOTc0MzMsIm5iZiI6MTczNTI5NzEzMywicGF0aCI6Ii82MDkxMjMyMC8zOTg3OTkzODYtY2JmMGFkNDAtYmE5Ny00NDQzLThlODctMjA3MjIxNGIxZmQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI3VDEwNTg1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5NmVjMmZiMjdiOTdjY2VlYWQ4OGFhMmZjZWY1MGIzYjNkOTY1ZTg3MjAxNWY3ZmNlZTg4ZGVjNGMzZTEwNDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.99gcg5O79el4ufemlGZeJ6RnutJD15JbwcHYKVnGuj4)

> Installing webkit2gtk4.0 and launch with WEBKIT_DISABLE_COMPOSITING_MODE=1 variable gets the startup wizard working, but only the menus work, the render & preview panes are blank.
> 
> Fedora 40 Gnome 46 (primarily Wayland) nVidia V2.1.1 Appimage with Gear Lever

I'm facing exactly the same problems as the other people above. Has anyone come up with a solution? Flatpak did not work for me.

kdave5, 2025-02-05:
This is still an issue on whatever desktop environment I run inside fedora 41. None of the fixes above seem to work.

saimkaan, 2025-02-10:
It works perfect on my Notebook without Nvidia drivers. When I try using it on my PC with Nvidia installed it wont show the preview pane.

witc-git, 2025-02-19:
I can get the AppImage (and flatpak beta) to work without any modification on my Intel laptop with integrated graphics running Fedora 41. On my Ryzen PC with NVidia GPU (official drivers) neither the AppImage or Flatpak beta work out of the box. For AppImage I got a blank startup wizard, environment variables got me past the startup wizard, but crashed the second I tried to load a file. For the flatpak 2.3.0 beta environment variable applied with Flatseal worked and left me with a functional slicer. I used the following variables in Flatseal.

`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`
`WEBKIT_DISABLE_COMPOSITING_MODE=1`

witc-git, 2025-03-03:
@Scout339 I found it on the releases page.

ongedotorg, 2025-03-05:
I gave up trying to make this work, the windows version with wine works flawlessly in Fedora 41

va7wv, 2025-05-11:
How does anyone make this software work on Linux?  I have to throw in the towel, it's just too buggy and frustrating.  Does anyone actually test it?