OrcaSlicer/OrcaSlicer #1842: Retraction Tower STL Non-manifold edges

DewiEJ, 2023-08-19:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

DewiEJ, 2023-08-19:
Slicer reports that the retraction_tower.stl from the calibration menu contains 53 non-manifold edges

Orca Slicer 1.6.4-beta2 Intel Mac
Prusa MK3S

[image]

A-A-Ron69420, 2023-09-18:
I'm getting the same error on 1.7 - beta of 40 non-manifold edges, does this actually affect anything???

JosefWN, 2023-10-03:
The STL file bundled with the application does not contain manifold edges but as it's cut, the cut version gets non-manifold edges.

apriendeau, 2023-12-19:
This is also occurring for me on 1.8.1.

Ramttid, 2024-01-23:
Still occuring

blindbatts, 2024-01-25:
Happening to me on 1.9.0 on macos

closed 2024-05-06 by github-actions[bot]

TheMoye, 2024-07-13:
Hi everyone, this error is still present in Orca Slicer 2.1.1 on Windows 11.

Ergonomicmike, 2024-08-03:
And on the 2.1.1 August 1 Nightly. It used to offer a Windows (10) repair tool to fix. But it's not offering that now.