OrcaSlicer/OrcaSlicer #1844: Change first layer height per-object for z-offset calibration

knw257, 2023-08-19:
**Is your feature request related to a problem? Please describe.**
Trying to run a z-offset test, but I cannot find any way to change the first layer height on a per-object basis

**Describe the solution you'd like**
When changing settings per object in the slicer, I'd like the "First Layer Height" option to be available to change

**Describe alternatives you've considered**
Only alternative currently is to create multiple different prints with the first layer height changed in each

**Additional context**
A Z-offset calibration test where a range of 10mm*10mm squares are printed with slightly different layer heights (e.g. -0.3 through +0.3mm) would also be helpful

richard-scott, 2024-01-05:
Hey, I just noticed that a global Z offset feature was added in v1.8.0:

[image]

closed 2024-04-13 by github-actions[bot]

typetetris, 2024-12-22:
can we reopen this? Would be a great feature!