OrcaSlicer/OrcaSlicer #1853: Temps in filament settings not being inserted into sliced file

poppinfresh69, 2023-08-19:
**Describe the bug**
User values entered in Print Temperatures in Filament Settings are not being automatically inserted into code of the sliced file. 

**3mf File for This Bug**
N/A

**To Reproduce**
Go to Filament Settings > Nozzle or High Temp Plate settings and change values. 
Slice file and view generated code under filament start G-code.
Notice no code is generated for the nozzle or bed temps inputted by the user.
Should have set bed and set extruder temp G-code commands
Only way to get code to appear is to enter manual commands in Advanced > Filament Start G-code 

**Expected behavior**
Filament start G-code should have nozzle and bed temperature commands auto created from user entered values in the UI instead of requiring manual G-code entries.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
N/A

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.6.4 Beta 3

Martorias, 2023-08-23:
I'm using klipper so via the Printer Machine G-code settings it starts with 
`print_start EXTRUDER=[first_layer_temperature] BED=[first_layer_bed_temperature] CHAMBER=[chamber_temperature]`
This works fine, but if I set first layer bed temp to 55 and then 60 it just starts at 55 and doesn't increase the next layer.

warnesj, 2023-08-24:
I was having the same problem in v1.6.4, but it looks like it's fixed now in v1.6.6.

warnesj, 2023-08-25:
I may have spoken too soon.  In v1.6.6 it appears the bed temperature gets set when I have Supports enabled, but when I disable them the bed temperature doesn't get set.  Looking at the generated G Code it looks like the M190 command is missing.

oleksii-suprun, 2023-10-08:
I have the same issue as @Martorias. I set the bed temp for the first level to 85 and 90 for other layers (according to the Prusa Slicer's default profile for PETG), but after the first layer, the temperature still stays at 85. 
[image]

oleksii-suprun, 2023-11-01:
Here is a small update on this issue. It looks like everything works well when both heater (nozzle and bed) temperatures are different for the first layer. For example, when I sed nozzle temperature to values 240/250 and bed 85/90 - everything works well. But when I set the nozzle to 240/240 and bed 85/90, I saw this issue happen and the bed temperature is not updated for subsequent layers and stays 85.

Greg767, 2024-02-08:
> Here is a small update on this issue. It looks like everything works well when both heater (nozzle and bed) temperatures are different for the first layer. For example, when I sed nozzle temperature [...]

I am seeing this issue as well although for first layers I want to set a higher temp value and when this is the case, it doesn't work. Temperatures are not changed from what is set in print start macro. So basically whatever temps I put in the macro that will be the temps no matter what is in the filament settings.

ohyeah8000, 2024-03-05:
Hey I still have this problem and my sliced file do not contain the temperature information. 

I found out about it after switching to Orcaslicer and hitting wrong temperatures (well the temps defined as default in my start_print macro). 

Where are the places I should look for solving the issue?

When I delete the defaults an error messages comes up. 

[image]

skyguynca, 2024-04-26:
Orca is not generating the first layer temp in the gcode for the extruder. I put it in manually in the print start macro but then in the layer change the m104 s208 command is ignored and the extruder temp goes to 0. It does this in every stl I slice

closed 2024-06-04 by SoftFever

Grumbledore667, 2024-12-18:
[image]
[image]

1st layer temp is still not added. Same goes for speeds. Set everything to 60ms/s, even for the infill. The infill was definetly way above 60ms/s.

micaheli, 2025-02-28:
I had this same issue. I used a built-in profile and it never generated.   I blew it away and started a profile from scratch and it now inserts the gcode.    This is definitely a bug, or that wouldn't happen.