OrcaSlicer/OrcaSlicer #1857: Default AnkerMake profile layer change G-code causes extrusion issues

staal54a, 2023-08-19:
**Describe the bug**
When running with the 1.6.4 Beta3 version on Windows, the default Anker M5 profile includes a default "Before layer change G-code" which is G92 E0 (resetting the extruder). This is problematic because the slicer already inserts a G92 E0 at the end of each layer but the real issue is that it seems "Retract at Layer Change" is always implicitly enabled regardless of the setting (I will be filing a separate bug later so if I forget, please remind me to link it).

What this means currently, is that the extruder position is reset to zero, then it attempts to retract. Since the extruder is all in absolute values for the default Anker profile, it means that instead of retracting, it actually extrudes the entire length used on the previous layer (minus the retract amount). This results in potentially massive blobs being formed on a print that scale depending on the footprint of each layer.

**To Reproduce**
1. Add the default Anker M5 printer profile (I use 0.4 mm nozzle and haven't tried it with others but I assume it's likely the same)
2. Don't change any settings and slice any object.
3. Go into the generated g-code (either export it or via the preview pane)
4. Search for the comment ";BEFORE_LAYER_CHANGE". See that there is a G92 E0 command immediately below that and then another one a few lines down as well

**Expected behavior**
I expect the slicer to handle any G92 E0 calls when in absolute mode and not require any custom G-code inputs in the printer profile. That "custom" code should be deleted.

**Screenshots**
See in the below screenshots after I sliced a simple cube, I marked the two G92 E0 commands at the beginning of a layer and the attempted retraction line
[image]

**Printer model**
AnkerMake M5

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 1.6.4 Beta3

staal54a, 2023-08-19:
The Retract at Layer Change bug I mentioned is #1858

staal54a, 2023-08-19:
I also forgot to mention I am specifically referring to the "Before layer change G-code" not the "Layer change G-code" though hopefully that is obvious if someone looks into it as the latter one is empty by default.

SoftFever, 2023-08-20:
Nice catch!

"G92 E0" instruction in layer change gcode indeed shouldn't be compatiable with absolute extruder.
I will add a warning if that happens.
Also change M5 to relative extruding

closed 2023-12-01 by github-actions[bot]

SX91, 2024-03-04:
Hello!

With unchecked relative extrusion and enabled retraction Orca Slicer still generates G92 E0 after retractions, making the G-Code unprintable on printers with no relative extrusion support.

G-Code flavor "Marlin (legacy)".

```
G1 X14.314 Y14.274 E18.62885
G1 E12.62885 F2700 ; <--- retraction
G92 E0 ; <--- distance reset here
G1 Z.7 F5760
G1 X14.14 Y-13.25 Z.7
G1 Z.3
M73 P2 R20
G1 E6 F2100
;TYPE:Bottom surface
;WIDTH:0.504556
G1 F2398.823
G1 X13.446 Y-13.944 E6.0539
```