OrcaSlicer/OrcaSlicer #1858: Can't disable Retract at Layer Change for Anker M5

staal54a, 2023-08-19:
**Describe the bug**
It seems that Retract at Layer Change is always implicitly enabled when slicing for the Anker M5, regardless of if the Retract at Layer Change box is ticked in the printer or filament settings.

**To Reproduce**
Use the default Anker M5 profiles for the printer and generic PLA which should mean Retract at Layer Change is disabled (it might be useful to enable the Verbose G-Code option but otherwise everything should be default). 

Slice any object like a cube then look through the generated G-code to find where a layer change occurs. Before and after the G1 command that moves to the new layer height, there are retract and then unretract G1 commands (unretract command on the default profile in this version will be G1 E1.5 F3600). Retract at Layer Change is disabled so this is incorrect.

**Expected behavior**
When Retract at Layer Change is disabled, there should be no retraction occurring at the layer change.

**Screenshots**
In the below screenshot I used all default Anker M5 profiles then enabled Verbose G-code. See where I marked the retract and unretract commands occurring at the layer change.
[image]

**Printer model**
AnkerMake M5

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 1.6.4 Beta3

staal54a, 2023-08-23:
I just tried slicing for my modified Creality CR-6 SE by setting up a custom Marlin printer and I see the same issue where, even if retract at layer change is disabled, there is still a retract and prime inserted at the beginning of each layer.

closed 2023-12-01 by github-actions[bot]

evilDave, 2024-06-14:
Can we re-open this issue (or shall I create another)?
I am unable to disable retract at layer change, no matter the setting, there is a retraction at layer change.

tiziano149, 2024-12-10:
Please let's reopen this, I have the same issue on the Bambu Lab A1 Mini!